VOLTTRON/volttron #1568: Tagging Error: RPC Call on add_tags

dmossie, 2017-11-29:
When I make an RPC call on the Mongo tagging service method "add_tags", I get this error:
volttron.platform.jsonrpc.RemoteError("TypeError('query_topics_by_pattern() takes exactly 1 argument (2 given)')")

Here is my code for where the error happens:
[image]

schandrika, 2017-12-14:
Could you please confirm if you are running the latest version of historian.  query_topics_by_pattern was added to historian sometime in July.  I am not able to reproduce the error when I test the add_tags api with latest historian.  Below is the code I used for a standalone test agent since it looks like you are testing from an agent outside of volttron. 

```
# Agents need access to VOLTTRON_HOME even if running in standalone mode
# to keep track of keys. This sets a default home.
set_home()
    
if __name__ == '__main__':
    try:
        print(remote_url())
        agent = Agent(address=remote_url(),
                      identity='standalone_testagent')
        task = gevent.spawn(agent.core.run)
        try:
            gevent.sleep(3)
            # test for invalid topic
            tags = {'SFU/WestMall/AHU1': {'ahu': True}}
            result = agent.vip.rpc.call('sqlite.tagging', 'add_tags',
                                       tags=tags).get(timeout=30)
            print result

            # insert some test topic and then test adding tags for that
            send = [{'topic': 'record/my_topic',
                     'headers': {u'Date': u'2017-12-14T04:54:03.486178Z'},
                     'message': 1}]
            result = agent.vip.rpc.call('platform.historian',
                                        'insert', send).get(timeout=10)
            gevent.sleep(2)
            print result

            # now add tags
            tags = {'record/my_topic': {'maxVal': 5}}
            result = agent.vip.rpc.call('sqlite.tagging', 'add_tags',
                                        tags=tags).get(timeout=30)
            print result

            # query for the tag
            result = agent.vip.rpc.call(
                'sqlite.tagging', 'get_tags_by_topic',
                topic_prefix='record/my_topic').get(timeout=30)
            print result
        finally:
            task.kill()
    except KeyboardInterrupt:
        pass
```

craig8, 2017-12-14:
@schandrika the latest would be releases/5.0rc correct...though if you mentioned in July develop should have it as well.

schandrika, 2017-12-14:
I am not able to reproduce the error on releases/5.0rc branch either.

schandrika, 2017-12-19:
Reproduced the error with crate historian. One of the new methods added for tagging was not implemented in crate. This was implemented in Mongo and Sql.  Will add a crate implementation.

closed 2018-06-08 by dmossie