VOLTTRON/volttron #1570: unhandled exception in ForwardHistorian

carlatpnl, 2017-12-07:
NOTE:  you must use 2 separate physical platforms, the problem does
		not exhibit if using loopback addresses, it is unclear if
		the problem exhibits when using a single physical address (non-loopback)
		but different ports.
		
have platform2 up and running
start ForwardHistorian on platform1 ("custom_topic_list": ["heartbeat/control"])
after ForwardHistorian receives a message to forward, shutdown platform2
the following error is produced:

2017-12-07 08:51:30,618 __main__ DEBUG: handled: 1 number of items
2017-12-07 08:51:30,619 __main__ DEBUG: Sending alert from the ForwardHistorian
2017-12-07 08:51:30,619 volttron.platform.messaging.health DEBUG: from_json {'status': 'BAD', 'last_updated': '2017-12-07T16:51:28.316644+00:00', 'context': 'Timeout occured'}
2017-12-07 08:51:30,621 volttron.platform.agent.base_historian ERROR: An unhandled exception occured while publishing.
Traceback (most recent call last):
  File "/home/volttron/volttron/volttron/platform/agent/base_historian.py", line 940, in _process_loop
    self.publish_to_historian(to_publish_list)
  File "/home/volttron/volttron/volttron/utils/docs.py", line 47, in f
    return self.mthd(obj, *args, **kwargs)
  File "/home/volttron/volttron/services/core/ForwardHistorian/forwarder/agent.py", line 403, in publish_to_historian
    status = Status.from_json(self.vip.health.get_status())
  File "/home/volttron/volttron/volttron/platform/messaging/health.py", line 121, in from_json
    cp = jsonapi.loads(data)
  File "/home/volttron/volttron/volttron/platform/agent/json.py", line 72, in loads
    return _loads(s, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/zmq/utils/jsonapi.py", line 56, in loads
    return jsonmod.loads(s, **kwargs)
  File "/home/volttron/volttron/env/lib/python2.7/site-packages/simplejson/__init__.py", line 516, in loads
    return _default_decoder.decode(s)
  File "/home/volttron/volttron/env/lib/python2.7/site-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/home/volttron/volttron/env/lib/python2.7/site-packages/simplejson/decoder.py", line 395, in raw_decode
    ord0 = ord(s[idx])
KeyError: 0
2017-12-07 08:51:30,631 volttron.platform.agent.base_historian DEBUG: Reading from/waiting for queue

kmonson, 2017-12-07:
What version of VOLTTRON are you using and when was the last time you did a pull?

carlatpnl, 2017-12-07:
Releases/5.0rc
Pulled an hour ago and verified I still could not connect
And the fix I put in for my agent, to use build_agent did not help

craig8, 2017-12-07:
@carlatpnl I will set up a couple machines in our cloud and do this test explicitly.

carlatpnl, 2017-12-07:
Please let me know if you are able to reproduce it.

craig8, 2017-12-07:
I was unable to reproduce this

After you upgraded to 5.0rc did you rebuild the forwarder agent?  When you run vctl status you should have forwarderhistorian4.0.

What is your operating system?

carlatpnl, 2017-12-08:
platform2 is a raspberry pi
platform1 is debian 9

carlatpnl, 2017-12-08:
i did not rebuild forwarder agent.
i did not use vctl
i did not install anything
i ran everything from the command line.

craig8, 2017-12-08:
Please rebuild the forwarder

Use the following from your volttron root directory

""""
python scripts/install-agent.py -s services/core/ForwardHistorian \
    -c pathtoyourconfig \
    -i identityname \
    --start \
    --force
""""
This will rebuild your agent for you without messing up any of the keys associated with the agents.  Please consider putting this in a script in a config directory under the main volttron root (this area is .gitignored).

I did my test against raspian x86 (debian 8)  I can spin up a debian 9 image, but I believe that reinstalling the agents in question will make this work successfully.

carlatpnl, 2017-12-09:
correction, platform1 is debian8, not 9

schandrika, 2020-03-11:
Closing since we are unable to reproduce this error and we did not get new follow up information. Please reopen if this is still an issue

closed 2020-03-11 by schandrika