VOLTTRON/volttron #1600: VolttronCenter Dashboard Not show in VOLTTRON 5.0 RC2

vishuPPatel, 2018-01-20:
here we setup volttron 4.1 vesrion this dashboard show below Like :

[image]

but we setup VOLTTRON 5.0 RC2 version this dashboard show below Like :

[image]

this is bug i request to volttron Term ...

Thanks!!!...

vishuPPatel, 2018-01-23:
I Enter the IP:8080/ui/index.html

this url send me Not Found

Can you help me how to get solution of this problem..

Thanks!!!....

craig8, 2018-01-23:
It would have helped if I tyhped the right address.  Instead of 'ui' put 'vc'.  So sorry for the mix up.

vishuPPatel, 2018-01-24:
OK ..... Igot it...

Thanks!!!...