VOLTTRON/volttron #1605: Make VC platform_driver_id configurable

jhaack, 2018-02-02:
Currently hardcoded, make this id configurable and potentially a  list. Kisensum has made this configurable to talk to their simulator driver, but if this were a list it could handle both the simulator and the real driver.

Kisensum will tie a PR to this issue.

rob-calvert, 2018-02-02:
The new VolttronCentralPlatform config parameter, platform-driver-id, is implemented in PR https://github.com/VOLTTRON/volttron/pull/1601. I'll look into making it a list of driver IDs, gathering device metrics from all drivers in the list of driver agents -- thanks for that idea.

rob-calvert, 2018-02-02:
I've pushed a VolttronCentralPlatform change that implements Jereme's suggestion. The parameter is now named "platform-driver-ids", and it holds a list of platform driver IDs. It defaults to ["platform.driver"].