VOLTTRON/volttron #1629: In the OpenADR VTN, don't auto-assign a VEN ID to a site if the site already has a VEN ID

rob-calvert, 2018-03-05:
In the OpenADR Virtual Top Node (VTN) server, /volttron-applications/kisensum/openadr, the web UI for defining/editing a site definition auto-assigns it a VEN ID. This auto-assignment currently happens during site editing as well as during site creation. Please change it so that a VEN ID gets auto-assigned only if the site doesn't already have a VEN ID.

rob-calvert, 2018-03-05:
@hillrnate committed a fix for this issue today in VOLTTRON / volttron-applications as part of (unmerged) pull request #54 (OpenADR VTN Server / Demand Response Application).

hlngo, 2018-03-06:
@rob-calvert I don't see it. Is the reference PR correct?

hlngo, 2018-03-06:
@rob-calvert Ok. That is in volttron-applications repos. 
@jhaack Do we have any plan to merge this [PR](https://github.com/VOLTTRON/volttron-applications/pull/54)

closed 2018-04-13 by craig8