VOLTTRON/volttron #1640: VirtualEnv URL change breaks bootstrap process

jhaack, 2018-03-22:
VirtualEnv URL has changed again breaking the bootstrap process. This has happened before #600. Fork their repo or find some other solution to prevent this changing out from underneath us again.

craig8, 2018-03-22:
The issue is caused because pypa/virtualenv hasn't released version 15.2.0 from their repo, however from https://pypi.python.org/pypi/virtualenv the latest reversion is 15.2.0.

The working around is modifying the get_version function to insert self.version="15.1.0" 

```` python
#Current
self.version = re.search(
    r'<revision>([^<]*)</revision>', doap_xml).group(1)
return self.version

# After
self.version = "15.1.0"
return self.version

````

nmerket, 2018-03-23:
I'm running into this issue as well. I see the workaround and will use it until they update, but this seems like an overly fragile way to create a virtualenv.

nmerket, 2018-03-23:
You could always pull it from PyPi directly, thereby avoiding the github/pypi version mismatch issue. Something like this:

```python
    f = urllib2.urlopen('https://pypi.python.org/pypi/virtualenv/json')
    r = json.load(f)
    if self.version is None:
        urls = r['urls']
    else:
        urls = r['releases'][self.version]
    tarball_url = None
    for url in urls:
        if url['packagetype'] == 'sdist':
            tarball_url = url['url']
            break
    assert(tarball_url is not None)
    self._fetch(tarball_url) # probably not quite right, but you get the idea
```

jhaack, 2018-03-23:
Another workaround if you have virtualenv on the system already:

```
virtualenv --prompt='(volttron)' env
. env/bin/activate
python setup.py -e .
```

nmerket, 2018-03-23:
For the record, what worked for me was

```
virtualenv --prompt='(volttron)' env
. env/bin/activate
pip install -e .
```

I still think downloading the package from pypi will be a more permanent solution to this problem.

commit 253991b7 (2018-03-23): Merge pull request #1641 from craig8/bootstrap-fix

Fixes #1640 Handle if virtualenv is not synced with version from pypa
commit a60369b3 (2018-03-23): Fixes #1640 Handle if the virtualenv is not synced with version from pypa.
commit 09a14dcb (2018-03-23): Fixed #1640
commit e6d51995 (2018-03-23): Fixed #1640
closed 2018-03-23 by craig8