VOLTTRON/volttron #1644: Can not install bootstrap.py in volttron4

TPponmat, 2018-03-26:
error is "raise HTTPError(req.get_full_url(), code, msg, hdrs, fp)
urllib2.HTTPError: HTTP Error 404: Not Found"

Can not access link   "https://github.com/pypa/virtualenv/archive/''{}.tar.gz"

craig8, 2018-03-26:
See issue #1640 

You say you are using volttron4.  From what url are you using?

TPponmat, 2018-03-26:
I clone it at url https://github.com/VOLTTRON/volttron.git
and after that 
sudo apt-get update
sudo apt-get install build-essential python-dev openssl libssl-dev libevent-dev git
git clone https://github.com/VOLTTRON/volttron
cd volttron

but after I run python bootstrap.py. >> Error is 
"raise HTTPError(req.get_full_url(), code, msg, hdrs, fp) urllib2.HTTPError: HTTP Error 404: Not Found"

craig8, 2018-03-26:
Thanks for that
 
You can use our release candidate by cloning using the following command

````
git clone https://github.com/VOLTTRON/volttron -b releases/5.0rc
````

The releases/5.0rc branch is far superior to the master branch.  In addition, it has the fix to the bootstrap that you are referencing from this issue.

TPponmat, 2018-03-27:
What version of ubuntu(14 or 16 or 17)  and python (2.7.???) to recommend for volttron eleases/5.0rc .

craig8, 2018-03-27:
We have used it on 12, 14, and 16 for ubuntu.  I am using it on mint 18 and it has been tested on centos as well.

craig8, 2018-04-13:
This issue has been resolved in master please pull the latest version

closed 2018-04-13 by craig8