VOLTTRON/volttron #1680: fix listener output

craig8, 2018-05-01:
### Description of Issue
Listener / log output does not output as expected

### Affected Version
noticed with python 2.7.12 and python 2.7.15r2 (ubuntu 18.04)
### Screenshots
[image]

#### Expected
Expected that the logger will resolve properly

### Steps to Reproduce

- create volttron
- install listener

### Additional Details

````
2018-05-01 14:20:08,956 (listeneragent-3.2 15287) listener.agent INFO: Peer: %r, Sender: %r:, Bus: %r, Topic: %r, Headers: %r, Message: 
%s
2018-05-01 14:20:08,972 (listeneragent-3.2 15287) listener.agent INFO: Peer: %r, Sender: %r:, Bus: %r, Topic: %r, Headers: %r, Message: 
%s
2018-05-01 14:20:12,332 (listeneragent-3.2 15287) listener.agent INFO: Peer: %r, Sender: %r:, Bus: %r, Topic: %r, Headers: %r, Message: 
%s
v2018-05-01 14:20:17,324 (listeneragent-3.2 15287) listener.agent INFO: Peer: %r, Sender: %r:, Bus: %r, Topic: %r, Headers: %r, Message: 
%s
2018-05-01 14:20:22,329 (listeneragent-3.2 15287) listener.agent INFO: Peer: %r, Sender: %r:, Bus: %r, Topic: %r, Headers: %r, Message: 
%s
````

kmonson, 2018-05-02:
I am unable to duplicate this on Python 2.7.12. I have tried running the agent directly on the command line (log to stderr) and through the platform and the log message appear correctly in each case.

kmonson, 2018-05-02:
Never mind. It doesn't affect master, only develop. Will investigate.

kmonson, 2018-05-02:
It seems to be a problem with the platform logging handler. It only does it if the log messages pass through the platform's logging system.

kmonson, 2018-05-02:
Here is the culprit: https://github.com/VOLTTRON/volttron/commit/9c5cff6b535ffb68436560e45756b25e629c350b#diff-fc3cfa63d7ad361a82096769b8b13ac3

PR #1682 (2018-05-02): Fix system logging
# Description

Fixes #1680 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
<a href='#crh-start'></a><a href='#crh-data-%7B%7D'></a>
[image]
[image]&nbsp;[image]
<a href='#crh-end'></a>
commit a0c62c16 (2018-05-02): Fixes #1680
commit 54dc9dd5 (2018-06-15): Fixes #1680