VOLTTRON/volttron #1691: Install volttron on Ubuntu Core  and Run As a system services fail

Soulweed, 2018-06-05:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue

ubuntu core don't use apt like ubuntu 16.04 then I try to use classic env for bringing apt back again
for install needed package such as python-dev etc. 
Now I can run volttron by manual command but when I try to config volttron.service and copy to path /etc/systemd/system successful the command  
'sudo systemctl start volttron.service' fail

craig8, 2018-06-05:
Have you checked out our https://github.com/VOLTTRON/volttron/blob/master/scripts/admin/volttron.service

We have successfully used this as a way of starting and stopping a volttron instance.

craig8, 2018-06-05:
ubuntu core...can you post me a link.