VOLTTRON/volttron #1713: YAML config loader chokes on tabs in JSON config files

acedrew, 2018-06-21:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue
A tab in a legacy json file is not parsed as valid by the YAML parser.

### Affected Version

Develop

### Screenshots

#### Expected

#### Actual

### Steps to Reproduce
Install agent with a JSON formatted config file that includes a tab character for readibility

### Additional Details

craig8, 2018-06-25:
@acedrew should fallback to the old way or search replace for /t and replace with spaces?  Thoughts?

acedrew, 2018-06-26:
I think the best option is just to cascade to the previous parser on error, without writing your own YAML/JSON parser, there's no way to know whether a tab is part of a field or in the whitespace/syntax.

craig8, 2018-06-26:
yup I agree...

PR #1735 (2018-07-10): Updated yaml config file parsing
# Description

This pull request updates the install-agent.py script to be more robust in error handling of parsing json as well as yaml configuration files.  It handles the \t parse error in yaml parser.

Fixes #1713, #1700 

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Tested against config file with tab characters in json file.  Testing with invalid json file and with yaml file.
<a href='#crh-start'></a><a href='#crh-data-%7B%7D'></a>
[image]
[image]&nbsp;[image]
<a href='#crh-end'></a>
commit c59ddf51 (2018-07-10): Fixed #1713 json loader with hardening of #1700
commit f7e26656 (2018-07-10): Fixes #1713 added pass through to json configuration routines
closed 2018-09-12 by craig8