VOLTTRON/volttron #1782: DNP3 Agent Address Issue

Kudrat9kaur, 2018-08-23:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue

### Affected Version

### Screenshots

#### Expected

#### Actual

### Steps to Reproduce

### Additional Details

I'm trying to run the DNP3 agent in volttron to communicate with a real device. When I specify the address of the remote device in the dnp3 config file, I get an error stating that the requested address can't be assigned. Could you help me with that? @rcalvert-cpi

Here's a screenshot of the logs:

[image]

rcalvert-cpi, 2018-08-23:
@Kudrat9kaur Based on how you've described this, I suspect that there may be a misunderstanding about the nature of VOLTTRON's DNP3Agent. It's not a device controller, it's an agent that acts as a DNP3 outstation, using TCP to communicate with a DNP3 master. So what you've described would only work, I think, if your "real device" is a TCP correspondent that's has implemented a DNP3 master. Please feel free to contact me directly if you'd like to have some further discussion about how you're trying to use DNP3 on VOLTTRON. -- Rob Calvert, rob.calvert@chargepoint.com

rcalvert-cpi, 2018-09-12:
After discussing this for awhile with him, I eventually sent this reply to @Kudrat9kaur:

The DNP3 outstation (the VOLTTRON agent) is a TCP server; the DNP3 master (your ABB unit) is a TCP client. The client initiates the TCP connection, and its configuration needs to know the IP address of the server, not the other way around. The VOLTTRON agent’s “local_ip” property should be the agent’s *own* IP address, not the address of the remote ABB unit.

By the way, if you haven’t already found the ReadTheDocs page for the VOLTTRON agent, https://volttron.readthedocs.io/en/latest/specifications/dnp3_agent.html, I recommend that you have a look at it.