VOLTTRON/volttron #1823: [Feature Request] Schedule agent functions with cron like schedules

dmossie, 2018-10-18:
When building an agent it would be awesome to schedule agent functions with cron like schedules.

Example:
```
@Core.cron('0 1 * * *')
def foo(self):
    ...

```

hashstat, 2018-10-18:
I think it would be nice to have a more generic decorator, or an overloaded `periodic` decorator, that takes an iterable of datetime objects.

Example:
```python
from volttron.platform.agent import cron

@Core.periodic(cron('0 1 * * *'))
def foo(self):
    ...
```
One could easily replace `cron()` with another generator, list, or other iterable to produce the desired schedule.

Perhaps it could also accept an iterable of floats/ints to produce more a more varied/customized periodic execution than what is currently possible.

hashstat, 2018-10-22:
I looked into this and found the schedule decorator/method to be a more appropriate candidate. In fact, it can be used to replace the periodic decorator/method as well.

```python
from volttron.platform.agent import cron
from volttron.platform.vip.agent import periodic

@Core.schedule(cron.schedule('0 1 * * *'))
def foo(self):
    ...

@Core.schedule(periodic(20))
def bar(self):
    ...
```

A patch is forthcoming.

craig8, 2018-10-25:
@dmossie I am going to close this as it is now in the develop branch.

PR #1827 (2018-10-24): Iterable schedule
# Description

Adds the ability to pass iterable objects to the core schedule decorator and associated method (dualmethod) as the deadline parameter. The iterable should yield datetime objects in increasing order and the scheduled function will be executed at the first time and automatically rescheduled at each execution, using the next time in the iterator. A new periodic() generator is added to use with the updated schedule functions, obsoleting the periodic decorator and associated method, which are now marked as deprecated.

Fixes #1823 

## Type of change

Please delete  [... cut]
closed 2018-10-25 by craig8