VOLTTRON/volttron #1831: Numpy Import Error

kl5pierce, 2018-10-29:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue

When attempting to run pytests for rabbitmq pubsub while in the ~/volttron directory, the test recieves ImportErrors, indicating that numpy needs to be imported, when test should be able to run without ImportErrors.

### Steps to Reproduce
```
cd ~/volttron
python bootstrap.py --testing
pytest -s -k test_granularity
```

### Additional Details

[additional_details_numpy_output.txt](https://github.com/VOLTTRON/volttron/files/2526763/additional_details_numpy_output.txt)

Temporary fix: `pip install numpy`

Note: if you go to the directory of the pytest:
`/home/username/volttron/volttrontesting/subsystems`

can run `pytest -s -k test_granularity` with no issues.

craig8, 2018-10-29:
what is test_granularity?  Where is it located?

kl5pierce, 2018-10-29:
/home/username/volttron/volttrontesting/subsystems in the rabbitmq-volttron branch.

craig8, 2018-10-29:
not finding it here https://github.com/VOLTTRON/volttron/tree/rabbitmq-volttron/volttrontesting/subsystems

kl5pierce, 2018-10-29:
My bad. It's one of the tests in https://github.com/VOLTTRON/volttron/blob/rabbitmq-volttron/volttrontesting/subsystems/test_rmq_pubsub.py

kl5pierce, 2018-11-26:
Quick note. The file that was initially causing the ImportError was located in: ~/volttron/services/core/OpenADRVenAgent/test/ControlAgentSim/controlagentsim/agent.py. However, the error can be fixed by modifying ~/volttron/optional_requirements.json

PR #1861 (2018-12-15): Fixed numpy import errors in pytests
# Description

Moves the base market tests to the proper location for platform specific services under volttrontesting/platform.  In addition the market tests will not run if numpy is not installed.  numpy is not a requirement for the main platform.

Fixes #1831 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Tested running all of the tests

**Test Configuration**:
# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have per [... cut]
closed 2018-12-20 by craig8