VOLTTRON/volttron #1834: VirtualEnv added src in their extraction script causing bootstrap to fail

craig8, 2018-10-31:
```vinod@vinod-ubuntu-vm:~/shared/volttron-repos/rabbitmq$ python bootstrap.py --rabbitmq
Creating virtual Python environment
Downloading virtualenv package information
Fetching https://pypi.python.org/pypi/virtualenv/json
latest release of virtualenv=16.1.0
Downloading virtualenv 16.1.0 directory /tmp/tmphpt4O0
Fetching https://pypi.python.org/packages/source/v/virtualenv/virtualenv-16.1.0.tar.gz
/usr/bin/python: can't open file '/tmp/tmphpt4O0/virtualenv-16.1.0/virtualenv.py': [Errno 2] No such file or directory```

But appending `src` directory to virtaulenv.py path, it worked.

```vinod@vinod-ubuntu-vm:~/shared/volttron-repos/rabbitmq$ python bootstrap.py --rabbitmq
Creating virtual Python environment
Downloading virtualenv package information
Fetching https://pypi.python.org/pypi/virtualenv/json
latest release of virtualenv=16.1.0
Downloading virtualenv 16.1.0 directory /tmp/tmpfT4uaI
Fetching https://pypi.python.org/packages/source/v/virtualenv/virtualenv-16.1.0.tar.gz
['/usr/bin/python', '/tmp/tmpfT4uaI/virtualenv-16.1.0/src/virtualenv.py', '--prompt', '(volttron)', '/home/vinod/shared/volttron-repos/rabbitmq/env']
New python executable in /home/vinod/shared/volttron-repos/rabbitmq/env/bin/python
Installing setuptools, pip, wheel...
done.
/home/vinod/shared/volttron-repos/rabbitmq/env/bin/python
Installing required packages
+ pip install wheel==0.30
Collecting wheel==0.30
  Using cached https://files.pythonhosted.org/packages/0c/80/16a85b47702a1f47a63c104c91abdd0a6704ee8ae3b4ce4afc49bc39f9d9/wheel-0.30.0-py2.py3-none-any.whl```

jhaack, 2018-11-01:
Manual fix is to edit bootstrap.py and make the following change to line 186 (depending on the code branch)

Change:
                 args.append(os.path.join(tmpdir, 'virtualenv-{}'.format(
                    self.version), 'virtualenv.py'))

To:
                    self.version), 'src/virtualenv.py'))

craig8, 2018-11-01:
This issue has now been fixed in the following branches

master, develop, rabbitmq-volttron

Sorry for the inconvenience

Craig

PR #1836 (2018-11-01): Fixes #1834 - develop
# Description

virtualenv 16.1.0 changes the package structure pin virtualenv to 16.0.0

Fixes #1834 

## Type of change

Please delete options that are not relevant.

- [x ] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Tested on linux mint
<a href='#crh-start'></a><a href='#crh-data-%7B%7D'></a>
[image]
[image]&nbsp;[image]
<a href='#crh-end'></a>
commit 20ad73a4 (2018-11-01): Fixes #1834
virtualenv 16.1.0 changes the package structure pin virtualenv to 16.0.0
commit e7e939fe (2018-11-01): Merge pull request #1836 from VOLTTRON/hotfix/issue1834

Fixes #1834 - develop
PR #1835 (2018-11-01): Fixes #1834
# Description

virtualenv 16.1.0 changes the package structure pin virtualenv to 16.0.0

Fixes #1834 

## Type of change

Please delete options that are not relevant.

- [x ] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Tested on linux mint
<a href='#crh-start'></a><a href='#crh-data-%7B%7D'></a>
[image]
[image]&nbsp;[image]
<a href='#crh-end'></a>
closed 2018-11-01 by craig8