VOLTTRON/volttron #1884: RabbitMQ-VOLTTRON: vctl status hangs if certificate of any service agent expires after successful volttron start

schandrika, 2019-01-17:
### Description of Issue
After a successful volttron startup if the service agent's certificate expires, vctl status command hangs indefinitely.  There are no error logs in rabbitmq or volttron
Killing the vctl status command shows the following on the console
(volttron)[vdev@cs_cbox myvolttron]$ vctl status
^XTraceback (most recent call last):
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/gevent/greenlet.py", line 534, in run
    result = self._run(*self.args, **self.kwargs)
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/grequests.py", line 72, in send
    self.url, **merged_kwargs)
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/requests/sessions.py", line 475, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/requests/sessions.py", line 596, in send
    r = adapter.send(request, **kwargs)
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/requests/packages/urllib3/connectionpool.py", line 595, in urlopen
    chunked=chunked)
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/requests/packages/urllib3/connectionpool.py", line 352, in _make_request
    self._validate_conn(conn)
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/requests/packages/urllib3/connectionpool.py", line 831, in _validate_conn
    conn.connect()
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/requests/packages/urllib3/connection.py", line 289, in connect
    ssl_version=resolved_ssl_version)
  File "/home/vdev/git/myvolttron/env/lib/python2.7/site-packages/requests/packages/urllib3/contrib/pyopenssl.py", line 350, in ssl_wrap_socket
    rd, _, _ = select.select([sock], [], [], sock.gettimeout())
KeyboardInterrupt
<Greenlet at 0x7fe5b80629b0: <bound method AsyncRequest.send of <grequests.AsyncRequest object at 0x7fe5b3fba110>>(stream=False)> failed with KeyboardInterrupt

### Affected Version
branch rabbitmq-volttron

#### Expected
vctl status or other volttron command should fail with error message or should timeout with exception going into log.

craig8, 2019-06-26:
@shwethanidd I don't see a referenced merge, however I do believe this has been addressed.  @schandrika  can you confirm and close if so.

schandrika, 2019-06-27:
It doesn't hang forever now, but still prints error on console and only quits after the configured time out period

(volttron)volttron@volttron1:~/git/myvolttron2$ vctl status
2019-06-27 14:34:01,149 () volttron.utils.rmq_mgmt ERROR: Unable to create RabbitMQ user for the agent. Check if RabbitMQ broker is running
Traceback (most recent call last):
  File "/home/volttron/git/myvolttron2/env/local/lib/python2.7/site-packages/gevent/greenlet.py", line 534, in run
    result = self._run(*self.args, **self.kwargs)
  File "/home/volttron/git/myvolttron2/volttron/platform/vip/agent/core.py", line 295, in run
    raise RuntimeError('VIP loop ended prematurely')
RuntimeError: VIP loop ended prematurely
<Greenlet at 0x7fec81b6c730: <bound method RMQCore.run of <volttron.platform.vip.agent.core.RMQCore object at 0x7fec81af0790>>(<gevent.event.Event object at 0x7fec81b25b90>)> failed with RuntimeError

status: operation timed out

craig8, 2019-06-27:
Great then closing this issue.

closed 2019-06-27 by craig8