VOLTTRON/volttron #2072: vctl status should not make an rpc call to all agents

craig8, 2019-07-10:
The current implementation of vctl status makes rpc function calls to each of the agents on the platform.  When there is a large number of agents on the platform this is not feasible and will timeout due to overloaded messagebus. 

A possible fix is to have the agent write it's health to a known location in the agent's directory and have the ControlService use that to get the status.

shwethanidd, 2019-07-10:
OR HealthMonitorService that does frequently polling to each agent to get health status and ControlService does a single RPC call to this agent to get status.

craig8, 2019-07-10:
@shwethanidd this is another path...Your suggestion has the added benefit of working for both installed as well as dynamic agents.