VOLTTRON/volttron #2076: error with auth

craig8, 2019-07-11:
```
2019-07-10 22:19:17,991 () volttron.platform.auth DEBUG: AUTH new capabilities update: {'listeneragent-3.3_1': [], 'b1362c8f-8f4d-4060-a02a-8b91a198930a': [], 'platform': []}
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
  File "/home/osboxes/repos/volttron-develop/volttron/platform/agent/utils.py", line 661, in watch_file
    callback()
  File "/home/osboxes/repos/volttron-develop/volttron/platform/auth.py", line 137, in read_auth_file
    self._send_update()
  File "/home/osboxes/repos/volttron-develop/volttron/platform/auth.py", line 170, in _send_update
    self._send_auth_update_to_pubsub()
  File "/home/osboxes/repos/volttron-develop/volttron/platform/auth.py", line 178, in _send_auth_update_to_pubsub
    frames = [zmq.Frame(b'auth_update'), zmq.Frame(json_msg)]
  File "zmq/backend/cython/message.pyx", line 141, in zmq.backend.cython.message.Frame.__cinit__
TypeError: Unicode objects not allowed. Only: str/bytes, buffer interfaces.
2019-07-11T02:19:18Z <Greenlet at 0x7fbc92059048: watch_file('/home/osboxes/.volttron/auth.json', <bound method AuthService.read_auth_file of <voltt)> failed with TypeError
```

dzimmanck, 2019-07-19:
String objects are unicode in Python3 by default, which makes the error message a little misleading.  Probably just need to encode the "json_msg".

frames = [zmq.Frame(b'auth_update'), zmq.Frame(json_msg.encode())]

craig8, 2019-07-20:
You are right that's exactly how it is now...though when I fixed it I didn't reference this issue :(

closed 2020-06-15 by craig8