VOLTTRON/volttron #2098: erroneous error in start up of historian

craig8, 2019-08-02:
Starting up the sqlhistorian (installed as platform historian using vcfg so it's configured as with sqlite) produces the following.  

```
2019-07-30 13:02:09,275 (sqlhistorianagent-3.7.0 4773) <stderr> ERROR: Generating LALR tables
2019-07-30 13:02:09,283 (sqlhistorianagent-3.7.0 4773) volttron.platform.agent.utils WARNING: WARNING: Agent id cannot be configured. It is a unique id assigned by VOLTTRON platform. Ignoring configured agentid
2019-07-30 13:02:09,402 (sqlhistorianagent-3.7.0 4773) volttron.platform.agent.base_historian ERROR: Trying to setup historian
```

The error does not effect the performance of the historian as data is freely flowing into the database.

I am not sure if this is the same for the other sql based historians or not but that would be good to check as well.

kmonson, 2019-08-06:
The first error is just output from the ply library. It's configured in the base historian to not write its generated parser to disk so you'll see it every startup. 

The second error will go away if you remove the "agent_id" setting from the configuration you are using. It's been long enough that we could probably remove that warning now. It was just to tell people that setting the agent id that way no longer works. You should probably remove the agent_id setting from all example configuration if it's not gone already.

The last one is an erroneous call to _log.exception. It should be _log.debug or _log info. You can find that here: https://github.com/VOLTTRON/volttron/blob/develop/volttron/platform/agent/base_historian.py#L1172

acedrew, 2020-06-21:
I'm getting non-deterministic behavior from an agent inheriting the base_historian as well, following the LALR error, it seems the historian just hangs with no output and no errors, after calling utils.load_config.
Strangely, logging the config output seems to fix the issue. Doing more testing now.

acedrew, 2020-06-21:
Does calling yacc.yacc(write_tables=False, debug=False) prevent the stderror output? could perhaps make debug an overridable default config so production systems don't have extraneous log output.