VOLTTRON/volttron #2099: Matlab documentation updates

craig8, 2019-08-02:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue

### Affected Version

### Screenshots

#### Expected

#### Actual

### Steps to Reproduce

### Additional Details

craig8, 2019-08-02:
@schandrika I disagree with the don't start the agent....the point is to use the config store of the agent to trigger the publish...not the starting of the agent.

schandrika, 2019-08-02:
In that case we should at least install and start without a config file.

craig8, 2019-08-02:
@schandrika Yes installing without a config file would be fine... @sgilbride  you can pass {} as the --config parameter so that it won't be required

craig8, 2020-06-15:
@sgilbride please verify this is complete and updated per our latest experience with new users.