VOLTTRON/volttron #2126: stop-volttron script fails to shutdown volttron platform

jklarson, 2019-10-09:
### Description of Issue
Running the stop-volttron script fails to shutdown the platform. There are no error messages in volttron.log, and the "closing connection to ZMQ" text displays as though it were shutting down, however the platform is still running.

### Affected Version
all-strings-python3 branch for 7.0 release

### Screenshots
[image]

#### Expected

#### Actual

### Steps to Reproduce
start volttron platform using zmq on all-strings-python3 branch

Have not tried RMQ, will try to reproduce using RMQ

schandrika, 2019-10-26:
Fixed bytes/string conversion and default user-id expected by main.py when trying to shutdown platform. This also took care of vcfg errors

shwethanidd, 2020-04-03:
@schandrika this is fixed right?

schandrika, 2020-04-03:
This looks like a timing issue. vctl shutdown --platform (and hence the wrapper stop-volttron) is not waiting till platform shutdown is complete.

closed 2020-04-03 by schandrika