VOLTTRON/volttron #2142: RMQ management console doesn't show up

craig8, 2019-11-08:


schandrika, 2019-11-08:
[image]

rjdkolb, 2019-12-11:
In the RabbitMQ installation in etc/rabbitmq/rabbitmq.conf

Remove lines for management.listener

E.g. :

#management.listener.ssl_opts.cacertfile = /home/myhome/.volttron/certificates/certs/mypc-trusted-cas.crt
#management.listener.ssl_opts.certfile = /home/myhome/.volttron/certificates/certs/mypc-server.crt
#management.listener.ssl_opts.keyfile = /home/myhome/.volttron/certificates/private/mypc-server.pem
The management interface is then accessible and you can login with a username and password.

shwethanidd, 2019-12-11:
@rjdkolb,
Those lines were added into rabbitmq.conf so that management interface uses our (VOLTTRON platform's) certificates. Yes, commenting those lines allows one to access the interface to login with username and password but that defeats the purpose that we are trying to achieve. The approach we need to take here is trying to add VOLTTRON platform's CA certificate into the browser. I'm looking at this issue now.

rjdkolb, 2019-12-11:
I understand now. I tried to install what I thought was the certificates as
a pkcs12 created by openssl (by combining the public and private keys) into
Firefox, but I failed to connect.

Fortunately we just wanted to look at the exchanges in RabbitMQ and see
what was going on under the hood when agents started and stopped. I will
also try again tomorrow.

Thanks.

craig8, 2020-01-10:
@rjdkolb @shwethanidd Just pinging to get a status on where this issue is currently at.

rjdkolb, 2020-01-10:
Hi @craig8 ,
I still have not figured it out, but the commenting out of the three lines is the work around that I have stuck with. This issue does not block anything for me for Volttron 7.

craig8, 2020-01-10:
Excellent.  @rjdkolb Please keep me up to date on issue you have found with 7 (aka develop now)

rjdkolb, 2020-01-10:
@craig8 , so far we are very happy with Volttron 7,RabbitMQ and Python 3.6 in develop. All our agents are running well over the last few weeks with minor changes to our code and we have done extensive tests with the PostgreSQL Historian. 

We don't use Volttron Central anymore, so we have not tested that.

Many thanks for your and the team's efforts.