VOLTTRON/volttron #2148: Test postgres historian with python3 in volttron-7-preview branch

schandrika, 2019-11-08:


rjdkolb, 2019-12-06:
The PostgreSQL historian has an issue with Python 3 and the DBDriver.
The DBDriver cursor no longer has a **.next()** and **.fetchone()** should be used.

I am preparing a pull request to fix this, but I need to fully test the Historian.

This is an example where both inserts work in Python 2, but only the "**.fetchone()**" method works in Python 3. i.e. using "**.fetchone()**" will not break Python 2.

```
import psycopg2
from psycopg2.sql import Identifier, SQL

from datetime import datetime

dbname = 'xxx'
user = 'user'
password = 'yourpass'

host = 'localhost'

conn = psycopg2.connect(dbname=dbname, user=user, password=password, host=host)
topics_table = 'topics'

with conn.cursor() as cursor:
    print('About to insert and get Primary key back with fetchone()[0]')
    random_string = datetime.today().strftime('%s') + '1'
    insert_topic_query = SQL('INSERT INTO {} (topic_name) VALUES (%(topic)s) RETURNING topic_id').format(Identifier(topics_table))
    cursor.execute(insert_topic_query, {'topic': random_string})
    print(cursor.fetchone()[0])
    print('DONE with insert and get Primary key back with fetchone()[0]')

with conn.cursor() as cursor:
    print('About to insert and get Primary key back with next()[0]')
    random_string = datetime.today().strftime('%s')  + '2'
    insert_topic_query = SQL('INSERT INTO {} (topic_name) VALUES (%(topic)s) RETURNING topic_id').format(Identifier(topics_table))
    cursor.execute(insert_topic_query, {'topic': random_string})
    try:
        print(cursor.next()[0])
        print('DONE with insert and get Primary key back with next()[0]')
    except Exception as e:
        print('Failed with insert and get Primary key back with next()[0] {e}'.format(e.message))
```

----
So this should change in "**postgresqlfuncts.py**"
```
    def insert_topic(self, topic):
        with self.cursor() as cursor:
            cursor.execute(self.insert_topic_query(), {'topic': topic})
            return cursor.fetchone()[0]

    def insert_agg_topic(self, topic, agg_type, agg_time_period):
        with self.cursor() as cursor:
            cursor.execute(self.insert_agg_topic_stmt(),
                           (topic, agg_type, agg_time_period))
            return cursor.fetchone()[0]
```

The same code is in "**redshiftfuncts.py**"

acedrew, 2019-12-08:
@rjdkolb Are you testing timescaleDB as well, or just vanilla postgres?

rjdkolb, 2019-12-09:
I am testing vanilla PostgreSQL and TimescaleDB at the same time.

So far so good except for this one issue.

rjdkolb, 2019-12-18:
@acedrew, I have completed basic testing on TimescaleDB and I have been working with a patched historian for over a week.
See https://github.com/VOLTTRON/volttron/pull/2179

rjdkolb, 2020-01-10:
We have completed our testing of the PostgreSQL Historian. It's working well on the develop branch for reading and writing of our datasets.

craig8, 2020-01-10:
@rjdkolb  Thanks!

closed 2020-01-10 by craig8