VOLTTRON/volttron #2158: VOLTTRON: bootstrap.py

rbindrabhatta, 2019-11-19:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue
I am unable to create a virtual environment to run VOLTTRON. Every time I run bootstrap.py, I get an error saying [SSL: CERTIFICATE_VERIFY_FAILED]. The screenshot is attached below. Is this the issue with the VOLTTRON version or do I need additional dependencies to run the file? Thank you. 
[image]

wolfwolf123, 2020-02-05:
This should be just and issue with the install elements. See this link:

https://stackoverflow.com/questions/45954528/pip-is-configured-with-locations-that-require-tls-ssl-however-the-ssl-module-in

Which should have instructions for whatever your os is.

craig8, 2020-06-15:
Closing as python3.6 and 3.7 not reproducible.

closed 2020-06-15 by craig8