VOLTTRON/volttron #2168: Discuss SQLHistorian Indexing

acedrew, 2019-12-01:
### Description of Issue
Discuss SQLHistorian Indexing

Indexes for the SQLHistorian are created on the timestamp in ascending order, does anyone know what the original reasoning for this was?
In most use cases, time series data is accessed most frequently from latest to earliest, meaning a descending order index will have to be scanned far less than an ascending index. 

In #2167 the TimescaleDB backend automatically creates a descending order index on time, so I disabled the default indexing and added a topic/time index which should increase performance substantially, especially given the way multitopic queries are currently constructed.

Obviously if the main workload is reading all the data from the beginning of time forward, then a descending index doesn't help much, however indexes only on time wouldn't be very useful either.

I'm happy to continue making improvements based on my use-cases and experience, but want to make sure I understand any other context that might have influenced this.

This is all defined in volttron/platform/dbutils/postgresqlfuncts.py in "setup_historian_tables"

### Affected Version

### Screenshots

#### Expected

#### Actual

### Steps to Reproduce

### Additional Details

rjdkolb, 2019-12-03:
Hi Andrew, 

I am also using the PostgreSQL historian with TimescaleDB.
I personally think this is actually a performance issue and it should be in descending order since you would usually show newer data.

Perhaps you can show some explain plans on the queries. (In the query function you can print the exact SQL for example)

We are enabling TimescaleDB on the data table using no special arguments. (No order for example)

Can you please share your experiences with Volttron and TimescaleDB?

Regards, 
Richard.

acedrew, 2019-12-03:
I've only done a little testing with VOLTTRON, enough to make the PR referenced. I have used TimescaleDB for a few other things, and am pretty familiar with time series data optimization.
I currently don't have enough data in an environment using TimescaleDB to get good benchmarking. I'd be happy to help though. 

One thing that definitely bears being tested is whether building and running individual queries for each topic is slower than just passing in a list of topics to query via the SQL. I think there may be some point where a long enough list my cause an issue, but I think that's true for either approach.

acedrew, 2019-12-03:
Another question I would have is are you using the Historian Agent to query data? Or just running queries against the DB separately?

craig8, 2019-12-03:
+1 on the multiple topic query

acedrew, 2020-06-23:
Still interested in this, I have about 150M records in TimescaleDB, through VOLTTRON. I've added a few different indexing options that seem to work well. I'm going to be updating my schema soon, and I have streaming replica setup for testing if folks have ideas.

craig8, 2020-06-24:
how is the streaming replica doing?

acedrew, 2020-06-24:
I should clarify, I'm using WAL-E file based replication, not actual WAL streaming. But for the data volumes we're dealing with, the latency is quite low. Also it has the added benefit of being a versioned backup in the S3 bucket it streams through.