VOLTTRON/volttron #2171: Bacpypes 0.17.7 breaks bacnet_scan and grab_bacnet_config in scripts in volttron-7-preview

jklarson, 2019-12-05:
### Description of Issue
After running bootstrap, bacpypes version 0.17.7 is installed, which features breaking changes for scripts/bacnet_scan.py and scripts/grab_bacnet_config.py

### Affected Version
volttron 7.0, volttron 7 preview branch

### Additional Details
Both scripts use bacpypes.service.device LocalDeviceObject which was removed in 0.17.7

kwarodom, 2019-12-07:
Same issue i faced yesterday when installing Volttron 7 for the first time since the preview version released. I installed Bacpypes 0.16.7 and it works as LocalDeviceObject was removed in 0.17.7 as you mentioned

shwethanidd, 2020-04-03:
@jklarson can you close the issue if resolved

commit c92bf110 (2020-01-29): Update requirements.py

Fix #2171 pinning bacpypes to 0.16.7
closed 2020-08-11 by craig8