VOLTTRON/volttron #2211: Segmentation fault for vcfg [Raspbian Stretch] [7.01rc1]

wolfwolf123, 2020-02-05:
I believe that this is an actual issue as I was able to setup the volttron platform successfully on an Ubuntu platform, but ran into a segmentation fault in running:

vcfg -- rabbitmq single
(See screenshot)

The Rabbit Status does report a error, but I'm under the impression that that is just because there is no server running at that point of the setup.

Running Raspbian Stretch (Fully Updated)
Freshly Installed OS
Freshly Installed Python3.7

On Tag 7.01rc1

### Screenshots
[image]
### Steps to Reproduce
Load Raspbian Stretch on Raspi
Install Python3.7
Run through steps in Readme found at https://github.com/VOLTTRON/volttron/tree/7.0rc1

Happy to provide log data, but I'm unsure where the vcfg program would be saving its log and I don't have any C debugging tools installed.

shwethanidd, 2020-02-05:
@laroque: Can you take a look at this?

laroque, 2020-02-05:
Building python 3.7.0 to try and reproduce now.

wolfwolf123, 2020-02-05:
If there is any other information I can provide to help reproduce I would be happy to. 

I realized that I ran into an SSL bug that I resolved by running:

sudo apt install libssl-dev libncurses5-dev libsqlite3-dev libreadline-dev libtk8.5 libgdm-dev libdb4o-cil-dev libpcap-dev

(https://stackoverflow.com/questions/45954528/pip-is-configured-with-locations-that-require-tls-ssl-however-the-ssl-module-in)

Which may have been part of the cause.

laroque, 2020-02-05:
@wolfwolf123 I've reproduced your problem, in fact it is a bit more generic: I find that on raspbian stretch and python 3.7.0; I get a segfault on any `vcfg` command; vctl fails with a more useful debugging info:

> ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found

(required by python3.7/site-packages/gevent/libuv/_corecffi.abi3.so) but 

> ➭ apt-cache madison libc6                                                                                            
>    libc6 | 2.24-11+deb9u4 | http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages

On Raspbian buster the libc6 is 2.28 and I don't see this same segfault (also on buster the native python3 package is 3.7.3 which should eliminate the need for installing python3 from source).

Is moving to buster an option in your case? I don't know that we have any confirmed cases of v7.0 on running on stretch and that may be easier than tracing possible version mismatch in system packages.

laroque, 2020-02-05:
In v7 with RMQ, I believe that the most likely supported Raspbian version will be buster. In order to get that installed I did have to manually install `libffi6`. You may also want to try installing from the develop branch, which includes changes staged for the next RC. More complete testing and documentation for running in this configuration is still a WIP.

craig8, 2020-06-15:
@laroque I think we have identified this and solved it.  Am I correct?

laroque, 2020-06-16:
Yes as far as I know this is fixed.

closed 2020-06-16 by craig8