VOLTTRON/volttron #2218: bootstrap.py --all

craig8, 2020-02-10:
### Description of Issue

Add a bootstrap.py --all to install all known requirements files for extra_requires in the platform.

schandrika, 2020-04-03:
Locking down versions of software in release,  is to communicate to the users what version of software this release was tested with. Optional requirements include packages that are needed by contributed code which we do not have the setup to test (For example, those requiring paid access to get an api key) I feel we shouldn't be locking version for these without knowing which version these codes were last tested with.  May be a good design going forward would be to move these out of core release and make them available separately. Thoughts?

shwethanidd, 2020-04-03:
Agreed. We can't locked down versions on those we haven't tested. With eclipse setup, we will have control over contributors. We need a plan to tackle that eventuality.

schandrika, 2020-04-03:
PR #2209 handles locking of versions for package in the current test environment. So closing this issue. We can create  a new issue to separate contributed code  that we cannot test into a different location once we move to eclipse

closed 2020-04-03 by schandrika