VOLTTRON/volttron #2220: Missing "Required Software"

JKolln, 2020-02-24:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue
these instructions (https://volttron.readthedocs.io/en/latest/setup/VOLTTRON-Prerequisites.html#volttron-prerequisites) are out of date.
The required packages shown are not adequate for the version of Linux I was setting up (Ubuntu 18.04.4). Here is the comparison:
 
sudo apt-get install build-essential python3.6-dev python3.6-venv openssl libssl-dev libevent-dev git

sudo apt-get install build-essential python3.6-dev python3.6-venv python3-venv openssl libssl-dev libevent-dev git libffi-dev

### VOLTTRON 7

### When using Ubuntu 18.04.4 LTS Server

### Additional Details
Failed bootstrapping until libffi-dev was installed

laroque, 2020-03-05:
These instructions are updated in #2219 (per automatic link from github above).

Question: @JKolln did you see this somewhere other than an RPi? I believe that libffi-dev is not required when running on Ubuntu 18 on x86.

Note 1:  On Ubuntu 18 the `python3-*` packages should be the same as the `python3.6-*` packages, you shouldn't need both. The updated instructions list only `python3-*` to be compatible with debian (which does not have `python3.6-*`). AFAIK 3.7 is also fine on distros where `python3-*` resolves to 3.7.

Note 2: I'm going to update the instructions to be less raspbian centric, since this solution (install libffi-dev) also works on Ubuntu.

shwethanidd, 2020-04-03:
@JKolln, @laroque: Is this still valid?

laroque, 2020-04-03:
AFAIK this was resolved in develop when #2219 was merged.

laroque, 2020-04-03:
Closing, @JKolln please let us know if there are still concerns.

closed 2020-04-03 by laroque