VOLTTRON/volttron #2230: Very slow interactions with ZMQ on RPi

laroque, 2020-03-04:
### Description of Issue
With an RPi running the develop branch using ZMQ, there have been two types of interactions that seem to be unreasonably slow:
1) following the `scripts/scalability-testing` folder, even with only 100 devices the timing tests take over 10 seconds
2) calls to `vctl status` take several seconds per agent

Repeating the same tests with RMQ do not have an issue, nor does the same test with ZMQ when run on VM on an x86 sysem.

### Affected Version
The develop branch (to be version 7)

### Steps to Reproduce
Install HEAD of develop on a raspberry Pi and configure with ZMQ and whatever basic set of agents (I have a master driver with 100 fake devices per the scalability-testing folder), as well as 5 ListenerAgents which are off.

### Additional Details
By adding extra logging, we've found that, at least in case 2 above, calls to `get_health` in `volttron/platform/control.py` are timing out in the call to `opts.connection.server.vip.peerlist().get(timeout=4)`, which means there is a 4 second delay any time that function call gets hit.

shwethanidd, 2020-11-18:
@laroque can you close this, since we don't see this issue anymore.

laroque, 2020-11-18:
yes, thanks

closed 2020-11-18 by laroque