VOLTTRON/volttron #2260: Unable to run volttron due to error in core.py

RichardLitt, 2020-03-26:
## Description of Issue

I am unable to run volttron due to a Nonetype object attribute error. Below, I replicate the error. I have to run `bootstrap.py` each time I start Volttron, or `volttron-pkg` is not found. This also seems odd, to me. I can never get to the `scripts/core/upgrade-listener` step in the README, as volttron fails to start adequately,

While nominally a support ticket, I want to log this here, because if the basic Setup docs don't work for a naive user, than they need to be changed. I'd be happy to work those changes in. 

I'm using Ubuntu 18.04.04 LTS.

```sh
12:23 ~/src/volttron (develop) 🐕  python3 bootstrap.py --force
WARNING: using non-empty environment directory: /home/richard/src/volttron/env
using non-empty environment directory: /home/richard/src/volttron/env
Installing required packages
+ pip install --upgrade wheel==0.30
Requirement already up-to-date: wheel==0.30 in ./env/lib/python3.6/site-packages
+ pip install --install-option --zmq=bundled --no-deps pyzmq
/home/richard/src/volttron/env/lib/python3.6/site-packages/pip/commands/install.py:212: UserWarning: Disabling all use of wheels due to the use of --build-options / --global-options / --install-options.
  cmdoptions.check_install_build_global(options)
Requirement already satisfied: pyzmq in ./env/lib/python3.6/site-packages
+ pip install --editable .
Obtaining file:///home/richard/src/volttron
Requirement already satisfied: PyYAML in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: cryptography==2.3 in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: gevent in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: grequests in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: ply in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: psutil in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: pyOpenSSL==19.0.0 in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: python-dateutil in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: pytz in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: pyzmq in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: requests in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: setuptools in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: tzlocal in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: watchdog-gevent in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: wheel==0.30 in ./env/lib/python3.6/site-packages (from volttron==7.0rc2)
Requirement already satisfied: cffi!=1.11.3,>=1.7 in ./env/lib/python3.6/site-packages (from cryptography==2.3->volttron==7.0rc2)
Requirement already satisfied: six>=1.4.1 in ./env/lib/python3.6/site-packages (from cryptography==2.3->volttron==7.0rc2)
Requirement already satisfied: idna>=2.1 in ./env/lib/python3.6/site-packages (from cryptography==2.3->volttron==7.0rc2)
Requirement already satisfied: asn1crypto>=0.21.0 in ./env/lib/python3.6/site-packages (from cryptography==2.3->volttron==7.0rc2)
Requirement already satisfied: greenlet>=0.4.14; platform_python_implementation == "CPython" in ./env/lib/python3.6/site-packages (from gevent->volttron==7.0rc2)
Requirement already satisfied: chardet<4,>=3.0.2 in ./env/lib/python3.6/site-packages (from requests->volttron==7.0rc2)
Requirement already satisfied: certifi>=2017.4.17 in ./env/lib/python3.6/site-packages (from requests->volttron==7.0rc2)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in ./env/lib/python3.6/site-packages (from requests->volttron==7.0rc2)
Requirement already satisfied: watchdog>=0.8 in ./env/lib/python3.6/site-packages (from watchdog-gevent->volttron==7.0rc2)
Requirement already satisfied: pycparser in ./env/lib/python3.6/site-packages (from cffi!=1.11.3,>=1.7->cryptography==2.3->volttron==7.0rc2)
Requirement already satisfied: pathtools>=0.1.1 in ./env/lib/python3.6/site-packages (from watchdog>=0.8->watchdog-gevent->volttron==7.0rc2)
Installing collected packages: volttron
  Found existing installation: volttron 7.0rc2
    Uninstalling volttron-7.0rc2:
      Successfully uninstalled volttron-7.0rc2
  Running setup.py develop for volttron
Successfully installed volttron
12:23 ~/src/volttron (develop) 🐕  source env/bin/activate
(volttron) 12:23 ~/src/volttron (develop) 🐕  volttron -vv -l volttron.log &
[1] 15346
(volttron) 12:24 ~/src/volttron (develop) 🐕  Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/richard/src/volttron/volttron/platform/main.py", line 843, in rmq_router
    stop(platform_shutdown=True)
  File "/home/richard/src/volttron/volttron/platform/vip/agent/core.py", line 539, in stop
    super(Core, self).stop(timeout=timeout)
  File "/home/richard/src/volttron/volttron/platform/vip/agent/core.py", line 321, in stop
    if gevent.get_hub() is self._stop_event.hub:
AttributeError: 'NoneType' object has no attribute 'hub'

Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
  File "/home/richard/src/volttron/volttron/platform/vip/agent/core.py", line 302, in run
    raise RuntimeError('VIP loop ended prematurely')
RuntimeError: VIP loop ended prematurely
2020-03-26T16:24:07Z <Greenlet at 0x7f974b748948: <bound method BasicCore.run of <volttron.platform.vip.agent.core.RMQCore object at 0x7f974b66ec50>>(<gevent._event.Event object at 0x7f974b62c468>)> failed with RuntimeError

Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
  File "/home/richard/src/volttron/volttron/platform/vip/agent/core.py", line 302, in run
    raise RuntimeError('VIP loop ended prematurely')
RuntimeError: VIP loop ended prematurely
2020-03-26T16:24:07Z <Greenlet at 0x7f974b748a48: <bound method BasicCore.run of <volttron.platform.vip.agent.core.RMQCore object at 0x7f974adf24e0>>(<gevent._event.Event object at 0x7f9749b05828>)> failed with RuntimeError

Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
  File "/home/richard/src/volttron/volttron/platform/vip/agent/core.py", line 302, in run
    raise RuntimeError('VIP loop ended prematurely')
RuntimeError: VIP loop ended prematurely
2020-03-26T16:24:07Z <Greenlet at 0x7f974b748e48: <bound method BasicCore.run of <volttron.platform.vip.agent.core.RMQCore object at 0x7f9749a9d7b8>>(<gevent._event.Event object at 0x7f9749ac8108>)> failed with RuntimeError

```

### Affected Version

`develop` branch.

### Screenshots

#### Expected

#### Actual

### Steps to Reproduce

### Additional Details

jklarson, 2020-03-26:
I am able to reproduce this issue on a clean Ubuntu 18.04. Running ./start-volttron can be used (after ending any already running volttron processes) while I debug this start command(I've tested this).

RichardLitt, 2020-03-26:
Awesome. Forgot to note: I also had to manually install `jwt` using pip (and had to install python3 pip). Might be worth adding this to the bootstrap file.

RichardLitt, 2020-03-26:
Running `./start-volttron` fails for me. 

``` 
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/richard/src/volttron/volttron/platform/main.py", line 843, in rmq_router
    stop(platform_shutdown=True)
  File "/home/richard/src/volttron/volttron/platform/vip/agent/core.py", line 539, in stop
    super(Core, self).stop(timeout=timeout)
  File "/home/richard/src/volttron/volttron/platform/vip/agent/core.py", line 321, in stop
    if gevent.get_hub() is self._stop_event.hub:
AttributeError: 'NoneType' object has no attribute 'hub'

e /home/richard/.volttron/auth.json loaded
2020-03-26 16:55:27,391 () volttron.platform.auth INFO: auth file /home/richard/.volttron/auth.json loaded
2020-03-26 16:55:27,391 () volttron.platform.auth INFO: loading auth file /home/richard/.volttron/auth.json
2020-03-26 16:55:27,391 () volttron.platform.auth INFO: loading auth file /home/richard/.volttron/auth.json
2020-03-26 16:55:27,392 () volttron.platform.auth INFO: auth file /home/richard/.volttron/auth.json loaded
2020-03-26 16:55:27,393 () volttron.platform.auth INFO: auth file /home/richard/.volttron/auth.json loaded
2020-03-26 16:55:29,337 () rmq_setup.py INFO: Rmq server at /home/richard/rabbitmq_server/rabbitmq_server-3.7.7 is running at 
2020-03-26 16:55:29,338 () root DEBUG: Creating RMQ Core config.store
2020-03-26 16:55:29,390 () volttron.platform.main ERROR: Unhandled exception in rmq router loop
Traceback (most recent call last):
  File "/home/richard/src/volttron/volttron/platform/main.py", line 835, in rmq_router
    bind_web_address=opts.bind_web_address
  File "/home/richard/src/volttron/volttron/platform/vip/rmq_router.py", line 95, in __init__
    param = self._build_connection_parameters()
  File "/home/richard/src/volttron/volttron/platform/vip/rmq_router.py", line 108, in _build_connection_parameters
    self._instance_name)
  File "/home/richard/src/volttron/volttron/utils/rmq_mgmt.py", line 996, in build_router_connection
    self.create_user_with_permissions(rmq_user, permissions, ssl_auth=self.is_ssl)
  File "/home/richard/src/volttron/volttron/utils/rmq_mgmt.py", line 907, in create_user_with_permissions
    if user not in self.get_users(ssl_auth):
  File "/home/richard/src/volttron/volttron/utils/rmq_mgmt.py", line 255, in get_users
    response = self._http_get_request(url, ssl_auth)
  File "/home/richard/src/volttron/volttron/utils/rmq_mgmt.py", line 171, in _http_get_request
    response = self._call_grequest('get', url, ssl_auth)
  File "/home/richard/src/volttron/volttron/utils/rmq_mgmt.py", line 123, in _call_grequest
    raise e
  File "/home/richard/src/volttron/volttron/utils/rmq_mgmt.py", line 111, in _call_grequest
    response[0].raise_for_status()
AttributeError: 'NoneType' object has no attribute 'raise_for_status'
```

craig8, 2020-04-01:
for the jwt you should use the different python bootstrap options...

python3 bootstrap.py --web --rabbitmq etc.

you can do this from an activated environment as well as to start the bootstrap process.

RichardLitt, 2020-04-02:
Thanks, @craig8. 

I've resolved these issues after a call with @jklarson. For now, closing this. If I run into them again, I'll reopen and submit a PR.

closed 2020-04-02 by RichardLitt