VOLTTRON/volttron #2270: Auth login failures

craig8, 2020-04-01:
### Description of Issue

Unable to authenticate using volttron login

From the develop branch with a brand new volttron environment run from python3 bootstrap.py --web.  

Add bind-web-address = http://127.0.0.1:8080 to the volttron config file which enables the web context.

Browse to localhost:8080 and add your user (I used admin for all three values)  Then attempt to login and it will fail.  Restart volttron and attempt to login and it will fail to login. 

I don't see anything suspicious in the log to suggest a total failure.

````
 v: <bound method AuthenticateEndpoints.get_auth_token of <volttron.platform.web.authenticate_endpoint.AuthenticateEndpoints object at 0x7f3ec8549cc0>>
2020-04-01 13:20:59,285 () volttron.platform.web.master_web_service DEBUG: registered route t is: callable
2020-04-01 13:20:59,285 () volttron.platform.web.authenticate_endpoint DEBUG: Username is: admin
2020-04-01 13:20:59,285 () passlib.handlers.argon2 DEBUG: detected 'argon2_cffi' backend, version '19.2.0', with support for 0x13 argon2 hashes
127.0.0.1 - - [2020-04-01 13:20:59] "POST /authenticate HTTP/1.1" 200 211 0.066420
2020-04-01 13:20:59,359 () volttron.platform.web.master_web_service DEBUG: path_info is: /admin/pending_auth_reqs.html
2020-04-01 13:20:59,359 () volttron.platform.web.master_web_service DEBUG: Peer path_info is associated with: None
2020-04-01 13:20:59,359 () volttron.platform.web.master_web_service DEBUG: MATCHED:
pattern: ^/admin.*, path_info: /admin/pending_auth_reqs.html
 v: <bound method AdminEndpoints.admin of <volttron.platform.web.admin_endpoints.AdminEndpoints object at 0x7f3ec8593748>>
2020-04-01 13:20:59,359 () volttron.platform.web.master_web_service DEBUG: registered route t is: callable
2020-04-01 13:20:59,359 () volttron.platform.web.admin_endpoints ERROR: Unauthorized user attempted to co
````
### Affected Version

develop branch

shwethanidd, 2020-04-01:
@sgilbride can you look at this?

craig8, 2020-04-02:
@sgilbride thanks!

PR #2272 (2020-04-02): Modified web to allow for http-based admin and vc.
Fixes #2270
closed 2020-04-02 by craig8