VOLTTRON/volttron #2281: VOLTTRON 6.0 installation and start platform error

geyin96, 2020-04-04:
When I installed and started VOLTTRON 6.0 platform, the error occurs:
[image]
And I found during the VOLTTRON installation, the only error is reported:
[image]
Then I reinstalled pyopenssl with compatible version of cryptography
`sudo pip install pyopenssl --upgrade cryptography`
however, the error still exists.
[image]

craig8, 2020-04-04:
The last image indicates that you have another instance running.  Please run 

````
ps aux |grep volttron

# Then kill the volttron 
kill procid
````

craig8, 2021-04-21:
assuming this is no longer an issue....

closed 2021-04-21 by craig8