VOLTTRON/volttron #2289: Example RabbitMQ Configuration file causes error

RichardLitt, 2020-04-07:
### Description of Issue

Setting up a Volttron instance following the develop branch README instructions causes a runtime error, if one uses the following command: 

```sh
vcfg --rabbitmq single examples/configurations/rabbitmq/rabbitmq_config.yml
# And then runs 
volttron -vv -l volttron.log &
```

### Affected Version

7.0

### Screenshots

See above.

#### Expected

No RuntimeError.

#### Actual

Error.

### Steps to Reproduce

Assuming you have an Ubuntu 18.04.3 instance:

```sh
# Update everything
sudo apt-get update -y
# Clone Volttron
git clone https://github.com/VOLTTRON/volttron --branch develop
cd volttron
# Set up RabbitMQ
./scripts/rabbit_dependencies.sh debian bionic
```

At this point, there's an undocumented step, which is necessary at least for Amazon EC2 instances:

```sh
sudo apt-get install python3-venv python3-pip
```

Then:

```sh
# Ensure that /etc/hosts has "127.0.0.1 localhost"
cat /etc/hosts | grep "127.0.0.1 localhost"
# Add some envvars for Rabbit
echo 'export RABBITMQ_HOME=$HOME/rabbitmq_server/rabbitmq_server-3.7.7'|sudo tee --append ~/.bashrc
source ~/.bashrc

# This is in the docs, but it will fail without starting the node.
$RABBITMQ_HOME/sbin/rabbitmqctl status
# So, start the node, and run it in the background.
# You can check on this after by running `$ jobs`.
$RABBITMQ_HOME/sbin/rabbitmq-server start &

# Active the venv
source env/bin/activate
# Configure Volttron
vcfg --rabbitmq single examples/configurations/rabbitmq/rabbitmq_config.yml

# Run Volttron
volttron -vv -l volttron.log &
```

At this point, the following error comes up: 

```
(volttron) ubuntu@ip-172-31-30-12:~/volttron$ volttron -vv -l volttron.log &
[1] 2334
(volttron) ubuntu@ip-172-31-30-12:~/volttron$ Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ubuntu/volttron/volttron/platform/main.py", line 850, in rmq_router
    stop(platform_shutdown=True)
  File "/home/ubuntu/volttron/volttron/platform/vip/agent/core.py", line 539, in stop
    super(Core, self).stop(timeout=timeout)
  File "/home/ubuntu/volttron/volttron/platform/vip/agent/core.py", line 321, in stop
    if gevent.get_hub() is self._stop_event.hub:
AttributeError: 'NoneType' object has no attribute 'hub'

Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
  File "/home/ubuntu/volttron/volttron/platform/vip/agent/core.py", line 302, in run
    raise RuntimeError('VIP loop ended prematurely')
RuntimeError: VIP loop ended prematurely
2020-04-07T17:34:45Z <Greenlet at 0x7fe7f3a83c48: <bound method BasicCore.run of <volttron.platform.vip.agent.core.RMQCore object at 0x7fe7f39e0198>>(<gevent._event.Event object at 0x7fe7f39e4ca8>)> failed with RuntimeError

Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
  File "/home/ubuntu/volttron/volttron/platform/vip/agent/core.py", line 302, in run
    raise RuntimeError('VIP loop ended prematurely')
RuntimeError: VIP loop ended prematurely
2020-04-07T17:34:45Z <Greenlet at 0x7fe7f3a83d48: <bound method BasicCore.run of <volttron.platform.vip.agent.core.RMQCore object at 0x7fe7f398a898>>(<gevent._event.Event object at 0x7fe7f220f588>)> failed with RuntimeError

Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
  File "/home/ubuntu/volttron/volttron/platform/vip/agent/core.py", line 302, in run
    raise RuntimeError('VIP loop ended prematurely')
RuntimeError: VIP loop ended prematurely
2020-04-07T17:34:45Z <Greenlet at 0x7fe7f221d348: <bound method BasicCore.run of <volttron.platform.vip.agent.core.RMQCore object at 0x7fe7f222b828>>(<gevent._event.Event object at 0x7fe7f220fd68>)> failed with RuntimeError
```

Nuking `.volttron/rabbitmq_config.yml` and starting over just using `vcfg --rabbitmq single` resolves the issue.

jklarson, 2020-04-08:
@RichardLitt I was able to reproduce the issue described above. This looks like a documentation item, which we'll be adding with the up-coming release. In the mean time, to use this example configuration, please update the entry for host in the example configuration with the fully qualified domain name of the system.

RichardLitt, 2020-04-09:
@jklarson Glad you were able to reproduce! This was one of the issues which confused me, which we sidestepped by not using it when we installed it together. Thanks.

PR #2298 (2020-04-08): Added clarification to readme for using example rabbitmq yml file
# Description

Adds documentation line describing an additional step necessary if using the example rabbitmq yml config file during platform setup.

Fixes #2289

This documentation addresses a problem that can occur when using the example rabbitmq yml config file when setting up a rabbitmq volttron instance using vcfg
closed 2020-06-15 by craig8