VOLTTRON/volttron #2317: NotFound Error While Trying to Access the Administration Area

gunduru, 2020-04-21:
### Description of Issue
Links for the Web GUI is broken for both https and http setup. Online documentation also mentiones /vc/index.html but this is not present either
https://localhost:8443/admin/login.html
http://localhost:8080/admin/login.html

### Affected Version
Latest

### Screenshots
[image]

#### Expected
A login page is expected

#### Actual
not found page

### Steps to Reproduce
1. Fresh install from github master
2. vcfg for a central instance
3. start volttron
4. reach default welcome page
5. click the administration area link and you get not found

### Additional Details

craig8, 2020-04-22:
@gunduru Please make sure you are using develop or the 7 pre release  not master as v7 hasn't been pushed to master as of yet.

gunduru, 2020-04-23:
Thanks a lot, just tried with releases/7.x branch and it's working now

closed 2020-04-23 by gunduru

craig8, 2020-04-24:
@gunduru Awesome! Glad it works for you!