VOLTTRON/volttron #2320: AsyncResult from the publish method gives timeout error

rkini-pnnl, 2020-04-22:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue
AsyncResults from the publish method returns "timeout" error.
For example,
line number 258 on datapublisher agent returns timeout error.
self.vip.pubsub.publish(peer='pubsub',
                                topic=all_topic,
                                message=message,  # [data, {'source': 'publisher3'}],
                                headers=headers).get(timeout=2) 

And, If you call the above function without the get, it publishes as usual.

The same problem you can reproduce with the ExternalData agent.
### Affected Version
Volttron 7.0 develop branch
### Screenshots
Following is the output from the DataPublisher agent 
___
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 854, in gevent._greenlet.Greenlet.run
  File "/home/roshan/Desktop/volttron/examples/DataPublisher/datapublisher/agent.py", line 334, in publish_loop
    self._publish_point_all(topic, message, publish_meta[topic], headers)
  File "/home/roshan/Desktop/volttron/examples/DataPublisher/datapublisher/agent.py", line 261, in _publish_point_all
    headers=headers).get(timeout=2)
  File "src/gevent/event.py", line 279, in gevent._event.AsyncResult.get
  File "src/gevent/event.py", line 304, in gevent._event.AsyncResult.get
  File "src/gevent/_abstract_linkable.py", line 178, in gevent.__abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 196, in gevent.__abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 182, in gevent.__abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent.__greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent.__greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent.__greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/__greenlet_primitives.pxd", line 35, in gevent.__greenlet_primitives._greenlet_switch
gevent.timeout.Timeout: 2 seconds
2020-04-22T22:35:29Z <Greenlet at 0x7f2e7ab72a70: <bound method Publisher.publish_loop of <__main__.Publisher object at 0x7f2e7ab1a150>>> failed with Timeout
___
If you remove ".get(timeout=2)" from the DataPublisher line 261, It will run as expected without the above error.

### Steps to Reproduce
Run DataPublisher or ExternalData agent with ubuntu 19.10 and fedora 31. This agent works as expected from lower ubuntu version.

### Additional Details

davidraker, 2020-04-23:
On OS that do not have this isssue (e.g. CentOS 8) the return from the AsyncResult.get() method called on the return from publish has also changed from VOLTTRON 6 to VOLTTRON 7:

On VOLTTRON 6 (Raspbian buster):
```
In [1]: from volttron.platform.vip.agent.utils import build_agent

In [2]: a = build_agent()
2020-04-23 01:44:35,028 volttron.platform.vip.agent.core DEBUG: address: ipc://@/home/volttron/.volttron/run/vip.socket
2020-04-23 01:44:35,030 volttron.platform.vip.agent.core DEBUG: identity: None
2020-04-23 01:44:35,032 volttron.platform.vip.agent.core DEBUG: agent_uuid: None
2020-04-23 01:44:35,033 volttron.platform.vip.agent.core DEBUG: serverkey: None
2020-04-23 01:44:35,283 volttron.platform.vip.agent.core INFO: Connected to platform: router: 89bcc200-054f-4870-a844-e44f378a1408 version: 1.0 identity: 75d6397f-37f7-4f27-be35-4fff462acd5c
2020-04-23 01:44:35,284 volttron.platform.vip.agent.core DEBUG: Running onstart methods.

In [3]: a.vip.pubsub.publish(peer='pubsub', topic='datalogger/Foo/Bar/Baz', mess
   ...: age=5).get()
Out[3]: '2'
```

On VOLTTRON 7 (CentOS 8):
```
In [1]: from volttron.platform.vip.agent.utils import build_agent

In [2]: a = build_agent()
2020-04-22 21:48:31,521 root DEBUG: Creating ZMQ Core None
2020-04-22 21:48:31,522 volttron.platform.vip.agent.core DEBUG: address: ipc://@/var/lib/volttron/.volttron/run/vip.socket
2020-04-22 21:48:31,522 volttron.platform.vip.agent.core DEBUG: identity: 7615f315-8352-4d7d-9d83-5f92ed0e83e0
2020-04-22 21:48:31,523 volttron.platform.vip.agent.core DEBUG: agent_uuid: None
2020-04-22 21:48:31,523 volttron.platform.vip.agent.core DEBUG: serverkey: None
2020-04-22 21:48:31,524 volttron.platform.vip.agent.core DEBUG: AGENT RUNNING on ZMQ Core 7615f315-8352-4d7d-9d83-5f92ed0e83e0
2020-04-22 21:48:31,526 volttron.platform.vip.zmq_connection DEBUG: ZMQ connection 7615f315-8352-4d7d-9d83-5f92ed0e83e0
2020-04-22 21:48:31,527 volttron.platform.vip.zmq_connection DEBUG: connecting to url ipc://@/var/lib/volttron/.volttron/run/vip.socket?publickey=VNVC33NjRLF-1IHYbe0YChPOgsY83iMA3FIz5dCgIFY&secretkey=lr3rW-ajxNv2zaSUJ9WzT-K8AofWAaqFu8qbP--rzqU&serverkey=VNVC33NjRLF-1IHYbe0YChPOgsY83iMA3FIz5dCgIFY
2020-04-22 21:48:31,527 volttron.platform.vip.zmq_connection DEBUG: url type is <class 'str'>
2020-04-22 21:48:31,552 volttron.platform.vip.agent.core INFO: Connected to platform: router: 3c7a5872-32b1-4525-ae2a-f5e01cd0aa08 version: 1.0 identity: 7615f315-8352-4d7d-9d83-5f92ed0e83e0
2020-04-22 21:48:31,552 volttron.platform.vip.agent.core DEBUG: Running onstart methods.

In [3]: a.vip.pubsub.publish(peer='pubsub', topic='datalogger/Foo/Bar/Baz', message=5).get()
Out[3]: '\x00\x00\x00\x00'
```

shwethanidd, 2020-04-28:
@davidraker: Thanks David, this is useful info. I need to check why integer value is not getting de-serialized properly.

shwethanidd, 2020-05-26:
Resolved in #2322 2331

closed 2020-05-26 by shwethanidd