VOLTTRON/volttron #2322: Both SQLHistorian agent and Forwarder can not create its sqlite database

rkini-pnnl, 2020-04-23:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue
Both the SQLHistorian and Forwarder agents can not create its SQLite database on Fedora 31 and Ubuntu 19.10.

### Affected Version
Volttron develop branch
releases/7.x

#### Expected

[image]

#### Actual
We get empty sqlhistorianagent-<version>-data directory, as shown below, for both forwarder and SQLHistorian agent:

[image]

### Steps to Reproduce
Install SQLHistorian agent with the default configuration on Ubuntu 19,10

UTGrid, 2020-04-23:
It is visible in the tree, but bears adding that the agent does successfully create a data directory when started. The historian.sqlite file, however is not added to the data directory after it is added.

davidraker, 2020-04-23:
It is visible in the tree, but bears adding that the agent does successfully create a data directory when started. The historian.sqlite file, however is not added to the data directory after it is added.

shwethanidd, 2020-05-26:
Resolved in #2346

closed 2020-05-26 by shwethanidd