VOLTTRON/volttron #2351: Base Historian sends pytz object to database in _caputre_log_data()

davidraker, 2020-05-23:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue
If meta['tz'] is provided in publish, it will be a string representation of the timezone. If that is not provided, for datalogger entries, it is extracted from the serialized datetime using:
```
my_ts, my_tz = process_timestamp(readings[0], topic)
```

my_tz is a pytz.timezone object which is then passed directly to the database driver. For sqlite, this raises an error which stops the historian from processing further publishes when it tries to put the entry into the backup.sqlite file.

### Affected Version
Probably all since August 2015.

### Screenshots

#### Expected
meta['tz'] should be a string, at least for saving to sqlite.

#### Actual
```
sqlite3.InterfaceError: Error binding parameter 3 - probably unsupported type.
```

### Steps to Reproduce
publish to datalogger topic without explicitly setting 'tz'.

### Additional Details
```
2020-05-23 13:41:34,706 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR: Exception in thread Thread-1:
2020-05-23 13:41:34,706 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR: Traceback (most recent call last):                                                                                                                                   
2020-05-23 13:41:34,706 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:   File "/usr/lib64/python3.8/threading.py", line 932, in _bootstrap_inner
2020-05-23 13:41:34,706 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:     self.run()
2020-05-23 13:41:34,706 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:   File "/usr/lib64/python3.8/threading.py", line 870, in run
2020-05-23 13:41:34,706 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:     self._target(*self._args, **self._kwargs)
2020-05-23 13:41:34,706 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:   File "/home/dmr/Projects/GridModerne/src/volttron/volttron/platform/agent/base_historian.py", line 1008, in _process_loop
2020-05-23 13:41:34,707 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:     self._do_process_loop()
2020-05-23 13:41:34,707 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:   File "/home/dmr/Projects/GridModerne/src/volttron/volttron/platform/agent/base_historian.py", line 1066, in _do_process_loop
2020-05-23 13:41:34,707 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:     cache_full = backupdb.backup_new_data((x for x in new_to_publish if x is not None))
2020-05-23 13:41:34,707 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:   File "/home/dmr/Projects/GridModerne/src/volttron/volttron/platform/agent/base_historian.py", line 1403, in backup_new_data
2020-05-23 13:41:34,707 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR:     c.execute('''INSERT OR REPLACE INTO metadata
2020-05-23 13:41:34,707 (sqlhistorianagent-3.7.0 250893) <stderr> ERROR: sqlite3.InterfaceError: Error binding parameter 3 - probably unsupported type.
```
This can be avoided by replacing:
```diff
-  meta['tz'] = my_tz 
+ meta['tz'] = my_tz.zone
```

craig8, 2020-05-26:
@davidraker thank you for this patch.  I will test the other historians and we can see if this doesn't break those.  doubtful, but better safe eh :)

davidraker, 2020-05-26:
Thanks. I didn't have the ability to test any others myself without a lot of setup, but it does look like all the datalogger publishes that do get stored in the local sqlite backup cache are getting stored as a string already. I didn't look into how the other historians pull the outstanding records to save, but it is always coming from the sqlite cache, your tests should (in principle) all pass. Let me know if there is anything I need to fix, though.

closed 2020-08-03 by shwethanidd