VOLTTRON/volttron #2386: Non-deterministic behavior in the MQTTHistorian

acedrew, 2020-06-21:
### Description of Issue
The MQTTHistorian fails to startup without an extraneous log call in the initialization.
The following code does *NOT* work:
`class MQTTHistorian(BaseHistorian):
    """This historian publishes data to MQTT.
    """

    def __init__(self, config_path, **kwargs):
        config = utils.load_config(config_path)

        # We pass every optional parameter to the MQTT library functions so they
        # default to the same values that paho uses as defaults.
        self.mqtt_qos = config.get("mqtt_qos", 0)
        self.mqtt_retain = config.get("mqtt_retain", False)`

Whereas this code *does* work:
`class MQTTHistorian(BaseHistorian):
    """This historian publishes data to MQTT.
    """

    def __init__(self, config_path, **kwargs):
        _log.debug(config_path)
        config = utils.load_config(config_path)
        _log.debug("this is just a message")

        # We pass every optional parameter to the MQTT library functions so they
        # default to the same values that paho uses as defaults.
        self.mqtt_qos = config.get("mqtt_qos", 0)
        self.mqtt_retain = config.get("mqtt_retain", False)
`

The failure mode will print the first debug message, then fail silently before completing the historian setup..

### Affected Version
7.0 under Python 3.8

#### Actual

### Steps to Reproduce

### Additional Details

acedrew, 2020-06-21:
It actually doesn't matter whether the log message prints or not, you can have a log output at debug level while running at WARNING+level and it will still cause a correct startup, vs failing with no log call.

craig8, 2020-08-11:
@acedrew now that 7.0 is finally out we may get some thoughts on this issue?

shwethanidd, 2020-09-09:
@acedrew: This agent works with latest develop  branch and on python 3.6.9. I'll try to reproduce in 3.8

shwethanidd, 2020-09-09:
@acedrew: I don't observe this behavior in 3.8 as well.

craig8, 2020-09-30:
Closing this because we couldn't reproduce...let us know if the issue comes up and we are able to reproduce it.

closed 2020-09-30 by craig8