VOLTTRON/volttron #2430: cannot start volttron process in the background with "volttron -vv" command

tpktang, 2020-08-10:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue

I followed your installation guide, when I run the "volttron -vv" command, I got this error. Before that, I run the "volttron -vv -l volttron.log&", I got error said "Missing library jwt within web package". So, I pip install jwt.
### Affected Version
7.1
### 
```(volttron) kwarodom@altonuc:~/workspace/volttron$ volttron -vv
2020-08-10 18:50:28,231 () volttron.platform.agent.utils DEBUG: value from env None
2020-08-10 18:50:28,231 () volttron.platform.agent.utils DEBUG: value from config False
2020-08-10 18:50:30,670 () rmq_setup.py INFO: Rmq server at /home/kwarodom/rabbitmq_server/rabbitmq_server-3.7.7 is running at 
2020-08-10 18:50:30,671 () root DEBUG: Creating RMQ Core config.store
2020-08-10 18:50:30,704 () volttron.platform.vip.rmq_connection DEBUG: ROUTER URL: <ConnectionParameters host=altonuc port=5671 virtual_host=volttron ssl=True>
2020-08-10 18:50:31,777 () volttron.utils.rmq_mgmt INFO: Creating ca signed certs for volttron1.config.store
2020-08-10 18:50:31,825 () volttron.platform.store INFO: Initializing configuration store service.
2020-08-10 18:50:31,826 () root DEBUG: Creating RMQ Core platform.auth
2020-08-10 18:50:31,832 () volttron.utils.rmq_mgmt INFO: Creating ca signed certs for volttron1.platform.auth
2020-08-10 18:50:31,853 () volttron.platform.auth INFO: loading auth file /home/kwarodom/.volttron/auth.json
2020-08-10 18:50:31,853 () volttron.platform.auth INFO: auth file /home/kwarodom/.volttron/auth.json loaded
2020-08-10 18:50:31,854 () volttron.platform.agent.utils INFO: Adding file watch for /home/kwarodom/.volttron/auth.json dirname=/home/kwarodom/.volttron, filename=auth.json
2020-08-10 18:50:31,855 () volttron.platform.agent.utils INFO: Added file watch for /home/kwarodom/.volttron/auth.json
2020-08-10 18:50:31,855 () volttron.platform.agent.utils INFO: Adding file watch for /home/kwarodom/.volttron/protected_topics.json dirname=/home/kwarodom/.volttron, filename=protected_topics.json
2020-08-10 18:50:31,855 () volttron.platform.agent.utils INFO: Added file watch for /home/kwarodom/.volttron/protected_topics.json
2020-08-10 18:50:31,855 () root DEBUG: Creating RMQ Core proxy_router
2020-08-10 18:50:31,859 () volttron.utils.rmq_mgmt INFO: Creating ca signed certs for volttron1.proxy_router
2020-08-10 18:50:33,714 () pika.adapters.base_connection ERROR: Read empty data, calling disconnect
2020-08-10 18:50:33,716 () pika.connection ERROR: Connection closed while authenticating indicating a probable authentication error
2020-08-10 18:50:33,716 () pika.connection ERROR: Connection setup failed due to Client was disconnected at a connection stage indicating a probable authentication error: (-1, 'EOF')
2020-08-10 18:50:33,718 () volttron.platform.vip.rmq_connection ERROR: Connection open error. Check if RabbitMQ broker is running.
2020-08-10 18:50:33,718 () volttron.platform.vip.rmq_connection ERROR: Connection closed unexpectedly, reopening in 30 seconds. router
2020-08-10 18:50:34,885 () pika.adapters.base_connection ERROR: Read empty data, calling disconnect
2020-08-10 18:50:34,887 () pika.connection ERROR: Connection closed while authenticating indicating a probable authentication error
2020-08-10 18:50:34,887 () pika.connection ERROR: Connection setup failed due to Client was disconnected at a connection stage indicating a probable authentication error: (-1, 'EOF')
2020-08-10 18:50:34,888 () volttron.platform.vip.rmq_connection ERROR: Connection open error. Check if RabbitMQ broker is running.
2020-08-10 18:50:34,888 () volttron.platform.vip.rmq_connection ERROR: Connection closed unexpectedly, reopening in 30 seconds. config.store
2020-08-10 18:50:34,889 () pika.adapters.base_connection ERROR: Read empty data, calling disconnect
2020-08-10 18:50:34,890 () pika.connection ERROR: Connection closed while authenticating indicating a probable authentication error
2020-08-10 18:50:34,890 () pika.connection ERROR: Connection setup failed due to Client was disconnected at a connection stage indicating a probable authentication error: (-1, 'EOF')
2020-08-10 18:50:34,890 () volttron.platform.vip.rmq_connection ERROR: Connection open error. Check if RabbitMQ broker is running.
2020-08-10 18:50:34,890 () volttron.platform.vip.rmq_connection ERROR: Connection closed unexpectedly, reopening in 30 seconds. platform.auth
2020-08-10 18:50:34,891 () pika.adapters.base_connection ERROR: Read empty data, calling disconnect
2020-08-10 18:50:34,891 () pika.connection ERROR: Connection closed while authenticating indicating a probable authentication error
2020-08-10 18:50:34,891 () pika.connection ERROR: Connection setup failed due to Client was disconnected at a connection stage indicating a probable authentication error: (-1, 'EOF')
2020-08-10 18:50:34,891 () volttron.platform.vip.rmq_connection ERROR: Connection open error. Check if RabbitMQ broker is running.
2020-08-10 18:50:34,892 () volttron.platform.vip.proxy_zmq_router DEBUG: ********************************************************************
2020-08-10 18:50:34,892 () volttron.platform.vip.proxy_zmq_router DEBUG: Stopping ZMQ Router
2020-08-10 18:50:34,892 () volttron.platform.vip.proxy_zmq_router DEBUG: ********************************************************************
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 854, in gevent._gevent_cgreenlet.Greenlet.run
  File "/home/kwarodom/workspace/volttron/volttron/platform/vip/proxy_zmq_router.py", line 163, in on_stop
    self.zmq_router.stop()
  File "/home/kwarodom/workspace/volttron/volttron/platform/vip/router.py", line 131, in stop
    self.socket.close(linger)
AttributeError: 'NoneType' object has no attribute 'close'
2020-08-10T11:50:34Z <Greenlet at 0x7f2150f30c48: <bound method ZMQProxyRouter.on_stop of <volttron.platform.vip.proxy_zmq_router.ZMQProxyRouter object at 0x7f21503ac9b0>>(<volttron.platform.vip.agent.core.RMQCore object a)> failed with AttributeError

2020-08-10 18:50:34,893 () volttron.platform.vip.rmq_connection ERROR: Connection closed unexpectedly, reopening in 30 seconds. proxy_router
2020-08-10 18:51:07,918 () pika.adapters.base_connection ERROR: Read empty data, calling disconnect
2020-08-10 18:51:07,919 () pika.connection ERROR: Connection closed while authenticating indicating a probable authentication error
2020-08-10 18:51:07,919 () pika.connection ERROR: Connection setup failed due to Client was disconnected at a connection stage indicating a probable authentication error: (-1, 'EOF')
2020-08-10 18:51:07,920 () volttron.platform.vip.rmq_connection ERROR: Connection open error. Check if RabbitMQ broker is running.
2020-08-10 18:51:07,920 () volttron.platform.vip.rmq_connection ERROR: Connection closed unexpectedly, reopening in 30 seconds. config.store
```

craig8, 2020-08-11:
@tpktang the missing jwt library can be ignored unless you are using the web platform.  From the trace it looks like you are using rabbitmq as your message bus.

Can you tell me about your deployment?  Was this an upgrade from a past system, give me the system specs, did you configure the system via vcfg or another issue?

craig8, 2020-08-11:
@tpktang the issue with the stdout is now fixed in develop see #2435