VOLTTRON/volttron #2499: rabbitmq_dependencies.sh

craig8, 2020-10-29:
### Description of Issue

At the end of running this script on a 20.04 system the following error occurs.

````
Note, selecting 'python-dev-is-python2' instead of 'python-dev'
Package libwxgtk3.0-0v5 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libwxgtk3.0-0v5' has no installation candidate
````

So the question is, is this something to be concerned with or just a differnce?

Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux

laroque, 2020-10-29:
looks like the apt package names have changed:

```
root@130c47f22a0e:/# apt-cache search wxgtk 
libwxgtk-media3.0-gtk3-0v5 - wxWidgets Cross-platform C++ GUI toolkit (GTK 3 media library runtime)
libwxgtk-media3.0-gtk3-dev - wxWidgets Cross-platform C++ GUI toolkit (GTK 3 media library development)
libwxgtk-webview3.0-gtk3-0v5 - wxWidgets Cross-platform C++ GUI toolkit (GTK 3 webview library runtime)
libwxgtk-webview3.0-gtk3-dev - wxWidgets Cross-platform C++ GUI toolkit (GTK 3 webview library development)
libwxgtk3.0-gtk3-0v5 - wxWidgets Cross-platform C++ GUI toolkit (GTK 3 runtime)
libwxgtk3.0-gtk3-dev - wxWidgets Cross-platform C++ GUI toolkit (GTK 3 development)
<snip>
```

My expectation would be that if installing dependencies fails, the rabbitmq_dependencies.sh script should have failed (so one way or another it is a problem I think)

shwethanidd, 2020-10-29:
@craig8: Can try changing the package name to 'libwxgtk3.0-gtk3-0v5' and rerunning the dependencies script