VOLTTRON/volttron #2508: VIP bytestring automatic handling not supported.

MachinesJesus, 2020-11-02:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue
VIP does not automagically handle bytestrings. Would be good to add this. 

### Affected Version
Using 7.0 but can test whatever you need. 

### Screenshots

#### Expected

#### Actual

### Steps to Reproduce
Send bytestring over VIP

### Additional Details

MachinesJesus, 2020-11-02:
Error Log:
2020-11-02 11:28:23,969 (awssagent-0.1 8076) <stderr> ERROR: Exception ignored in: <class 'SystemExit'>
2020-11-02 11:28:23,969 (awssagent-0.1 8076) <stderr> ERROR: Traceback (most recent call last):
2020-11-02 11:28:23,969 (awssagent-0.1 8076) <stderr> ERROR:   File "/home/ubuntu/te/vtrn/env/lib/python3.8/site-packages/awscrt/mqtt.py", line 477, in callback_wrapper
2020-11-02 11:28:23,969 (awssagent-0.1 8076) <stderr> ERROR:     callback(topic=topic, payload=payload)
2020-11-02 11:28:23,969 (awssagent-0.1 8076) <stderr> ERROR:   File "/home/ubuntu/.volttron/agents/8f65a133-2c14-45b5-a701-fda9b7e956bc/awssagent-0.1/awss/agent.py", line 173, in _receive_awsiot_message
2020-11-02 11:28:23,969 (awssagent-0.1 8076) <stderr> ERROR:     self.vip.pubsub.publish('pubsub', topic=topic, message=payload)  # See note below.
2020-11-02 11:28:23,969 (awssagent-0.1 8076) <stderr> ERROR:   File "/home/ubuntu/te/vtrn/volttron/platform/vip/agent/subsystems/pubsub.py", line 641, in publish
2020-11-02 11:28:23,969 (awssagent-0.1 8076) <stderr> ERROR:     self.vip_socket.send_vip('', 'pubsub', args, result.ident, copy=False)
2020-11-02 11:28:23,970 (awssagent-0.1 8076) <stderr> ERROR:   File "/home/ubuntu/te/vtrn/volttron/platform/vip/socket.py", line 471, in send_vip
2020-11-02 11:28:23,970 (awssagent-0.1 8076) <stderr> ERROR:     send(args, flags=flags, copy=copy, track=track)
2020-11-02 11:28:23,970 (awssagent-0.1 8076) <stderr> ERROR:   File "/home/ubuntu/te/vtrn/volttron/platform/vip/socket.py", line 415, in send_multipart
2020-11-02 11:28:23,970 (awssagent-0.1 8076) <stderr> ERROR:     parts = serialize_frames(msg_parts)
2020-11-02 11:28:23,970 (awssagent-0.1 8076) <stderr> ERROR:   File "/home/ubuntu/te/vtrn/volttron/utils/frame_serialization.py", line 109, in serialize_frames
2020-11-02 11:28:23,970 (awssagent-0.1 8076) <stderr> ERROR:     sys.exit(0)
2020-11-02 11:28:23,970 (awssagent-0.1 8076) <stderr> ERROR: SystemExit: 0

MachinesJesus, 2020-11-02:
Sure I can handle this but I think volttron should / could detect bytestrings and support via VIP or just convert, so volttron does not break. 
Why? Encoding / Secrecy and Other packages may return bytestrings.

craig8, 2020-11-02:
This should be fixed in the latest develop codebase.  

One should be able to pass anything serializable through the bus.

MachinesJesus, 2020-11-02:
Cool cool. OK will pull down the latest and test it.

craig8, 2020-11-02:
You could use main or develop...both should work

MachinesJesus, 2020-11-11:
@craig8 tried 8.0rc1 and develop. Both fail to pass our Bytes. Would be great to get this working. The issue seems to be on the frame breakup. 

"2020-11-11 10:55:07,725 (awssagent-0.1 13580) <stderr> ERROR:   File "/home/ubuntu/te/vtrn/volttron/utils/frame_serialization.py", line 109, in serialize_frames"

Thoughts:
Do you breakup the messages? _Appears to expect a List._ 
Into what size? Because I could potentially build this into protobuff.  _Appears to have no size limits_

Attempted to alter the fram_serialization to 'utf-8' this causes an error on re-install. 

2020-11-11 11:02:40,223 () volttron.utils.frame_serialization ERROR: Unicode decode error: 'utf-8' codec can't decode byte 0xbf in position 2: invalid start byte

craig8, 2020-11-11:
@MachinesJesus can you reach out to me on slack so we can schedule a time...I would love to chat more about the protobufs you mention here.  I have been looking into how I would utilize those for this type of issue.  Maybe you have some insight for me?

MachinesJesus, 2020-11-11:
@craig8 sent you a slack message. 

For any users who want to use AWS and protobuff. Do this:
Use ToJson() at all levels of protobuff serialization (just top level does not work), then x.decode("utf-8") once message received from awsiot. This will work.