VOLTTRON/volttron #2519: Bacnet Scan doesn't appear to work

gnmerritt, 2020-11-04:
### Description of Issue

I have a two-node volttron deployment, with a cloud node running VCP and an edge node running a bacnet proxy. I'm unable to complete a bacnet scan despite seeing IAM responses / devices show up in the logs on the edge node. I've traced the code through up to `self._pub_to_vc` in `vcplatform/agent.py` and am not sure where to look next for problems. Logs attached below for the central and edge nodes and I'm also happy to grab tcpdumps if they'd be helpful. 

### Affected Version

Seems to affect both 7.0.0 and the tip of `develop` (1ed2ee1577eb57232590101d0229b56bb36504eb committed on 11/4/2020)

### Screenshots

[image]

#### Expected

I'd expect to see a device show up in the UI. With the `bacnet_scan.py` script I get:

```
Device Address        = <Address 10.10.220.101>
Device Id             = 1
maxAPDULengthAccepted = 1024
segmentationSupported = segmentedBoth
vendorID              = 5
```

#### Actual

UI wheel spins forever

### Steps to Reproduce

Start a scan with parameters from the screenshot

### Additional Details

Logs from the central instance:

```
2020-11-04 21:16:23,452 () volttron.platform.web.master_web_service DEBUG: Peer path_info is associated with: volttron.central
2020-11-04 21:16:23,452 () volttron.platform.web.master_web_service DEBUG: Calling peer volttron.central back with env={'HTTP_USER_AGENT': 'Mozilla/5.0 (Macintosh; Intel Mac OS X 10_14_6) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.111 Safari/537.36', 'PATH_INFO': '/vc/jsonrpc', 'QUERY_STRING': '', 'REQUEST_METHOD': 'POST', 'SERVER_PROTOCOL': 'HTTP/1.1', 'REMOTE_ADDR': '172.31.87.61', 'HTTP_ACCEPT_ENCODING': 'gzip, deflate', 'HTTP_COOKIE': '_ga=GA1.2.1949159287.1603123476', 'CONTENT_TYPE': 'application/json', 'SERVER_NAME': '507f05e9eb60', 'wsgi.url_scheme': 'http', 'HTTP_HOST': 'c1.vc-dev.onboarddata.io:8080'} data={'method': 'start_bacnet_scan', 'authorization': 'eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJncm91cHMiOlsiYWRtaW4iXX0.5veb_4rYxk6Yf4mjHiFr4qmpf_zBapYvhDOAZTnyFuw', 'params': {'proxy_identity': 'platform.bacnet_proxy', 'platform_uuid': 'b2ItZWRnZTEucGxhdGZvcm0uYWdlbnQ=', 'target_address': '10.10.220.101', 'scan_length': '60'}, 'jsonrpc': '2.0', 'id': 'fda62b60-1ee2-11eb-9d1c-53557f568553'}
2020-11-04 21:16:23,453 (volttroncentralagent-5.2 304) volttron.platform.vip.agent.subsystems.web DEBUG: Calling function: jsonrpc
2020-11-04 21:16:23,454 (volttroncentralagent-5.2 304) __main__ INFO: rpc method: start_bacnet_scan
2020-11-04 21:16:23,454 (volttroncentralagent-5.2 304) __main__ DEBUG: REMOTE_ADDR: 172.31.87.61
2020-11-04 21:16:23,454 (volttroncentralagent-5.2 304) __main__ DEBUG: SESSION_USER IS: {'user': 'vt_admin', 'token': 'eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJncm91cHMiOlsiYWRtaW4iXX0.5veb_4rYxk6Yf4mjHiFr4qmpf_zBapYvhDOAZTnyFuw', 'ip': '172.31.87.61', 'groups': ['admin']}
2020-11-04 21:16:23,454 (volttroncentralagent-5.2 304) __main__ DEBUG: RPC METHOD IS: start_bacnet_scan
2020-11-04 21:16:23,454 (volttroncentralagent-5.2 304) __main__ DEBUG: inside _route_request fda62b60-1ee2-11eb-9d1c-53557f568553, start_bacnet_scan, {'proxy_identity': 'platform.bacnet_proxy', 'platform_uuid': 'b2ItZWRnZTEucGxhdGZvcm0uYWdlbnQ=', 'target_address': '10.10.220.101', 'scan_length': '60'}
2020-11-04 21:16:23,456 () volttron.platform.web.master_web_service DEBUG: Sending data to /vc/ws/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJncm91cHMiOlsiYWRtaW4iXX0.5veb_4rYxk6Yf4mjHiFr4qmpf_zBapYvhDOAZTnyFuw/management with message {"type": "start_bacnet_scan", "data": "{}"}
2020-11-04 21:16:23,457 () volttron.platform.web.webapp DEBUG: Sending message to clients!
2020-11-04 21:16:23,457 () volttron.platform.web.webapp DEBUG: Sending endpoint&&message /vc/ws/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJncm91cHMiOlsiYWRtaW4iXX0.5veb_4rYxk6Yf4mjHiFr4qmpf_zBapYvhDOAZTnyFuw/management&&{"type": "start_bacnet_scan", "data": "{}"}
2020-11-04 21:16:23,458 (volttroncentralagent-5.2 304) __main__ DEBUG: Calling start_bacnet_scan on platform b2ItZWRnZTEucGxhdGZvcm0uYWdlbnQ=
2020-11-04 21:16:23,458 (volttroncentralagent-5.2 304) Platforms DEBUG: Currently monitoring platforms
2020-11-04 21:16:23,458 (volttroncentralagent-5.2 304) Platforms DEBUG: vip: tcp://0.0.0.0:22916, identity: ob-collector.platform.agent
2020-11-04 21:16:23,458 (volttroncentralagent-5.2 304) Platforms DEBUG: vip: tcp://0.0.0.0:22916, identity: ob-edge1.platform.agent
2020-11-04 21:16:23,458 (volttroncentralagent-5.2 304) __main__ DEBUG: Handling bacnet_scan platform: b2ItZWRnZTEucGxhdGZvcm0uYWdlbnQ=
2020-11-04 21:16:23,459 (volttroncentralagent-5.2 304) Platforms DEBUG: Currently monitoring platforms
2020-11-04 21:16:23,459 (volttroncentralagent-5.2 304) Platforms DEBUG: vip: tcp://0.0.0.0:22916, identity: ob-collector.platform.agent
2020-11-04 21:16:23,459 (volttroncentralagent-5.2 304) Platforms DEBUG: vip: tcp://0.0.0.0:22916, identity: ob-edge1.platform.agent
2020-11-04 21:16:23,459 (volttroncentralagent-5.2 304) volttron.platform.vip.agent.subsystems.web INFO: Agent registering websocket at: /vc/ws/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJncm91cHMiOlsiYWRtaW4iXX0.5veb_4rYxk6Yf4mjHiFr4qmpf_zBapYvhDOAZTnyFuw/iam
2020-11-04 21:16:23,460 () volttron.platform.web.master_web_service DEBUG: Caller identity: volttron.central
2020-11-04 21:16:23,460 () volttron.platform.web.master_web_service DEBUG: REGISTERING ENDPOINT: /vc/ws/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJncm91cHMiOlsiYWRtaW4iXX0.5veb_4rYxk6Yf4mjHiFr4qmpf_zBapYvhDOAZTnyFuw/iam
2020-11-04 21:16:23,463 () volttron.platform.web.master_web_service DEBUG: res is a dictionary.
172.31.87.61 - - [2020-11-04 21:16:23] "POST /vc/jsonrpc HTTP/1.1" 200 188 0.011873
2020-11-04 21:16:23,656 () ws4py INFO: Managing websocket [Local => 172.17.0.2:8080 | Remote => 172.31.87.61:60489]
2020-11-04 21:16:23,657 () volttron.platform.web.websocket INFO: Socket opened
2020-11-04 21:16:23,658 () volttron.platform.web.webapp DEBUG: IDENTITY,CLIENT: volttron.central added to endpoint set
2020-11-04 21:16:23,659 (volttroncentralagent-5.2 304) volttron.platform.vip.agent.subsystems.web INFO: Client opened websocket ip: 172.31.87.61 endpoint: /vc/ws/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJncm91cHMiOlsiYWRtaW4iXX0.5veb_4rYxk6Yf4mjHiFr4qmpf_zBapYvhDOAZTnyFuw/iam
2020-11-04 21:16:23,659 (volttroncentralagent-5.2 304) __main__ DEBUG: OPENED ip: 172.31.87.61 endpoint: /vc/ws/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJncm91cHMiOlsiYWRtaW4iXX0.5veb_4rYxk6Yf4mjHiFr4qmpf_zBapYvhDOAZTnyFuw/iam
2020-11-04 21:16:23,660 (volttroncentralagent-5.2 304) __main__ DEBUG: Websocket allowed.
2020-11-04 21:16:27,951 (volttroncentralagent-5.2 304) __main__ DEBUG: Connected: {'ob-collector.platform.agent', 'ob-edge1.platform.agent'}
2020-11-04 21:16:37,959 (volttroncentralagent-5.2 304) __main__ DEBUG: Connected: {'ob-collector.platform.agent', 'ob-edge1.platform.agent'}
2020-11-04 21:16:41,139 (sqlhistorianagent-3.7.0 302) volttron.platform.agent.base_historian DEBUG: Queue wait timed out. Falling out.
[... 12 lines cut ...]
2020-11-04 21:16:25,482 (vcplatformagent-4.8 310) __main__ INFO: Starting bacnet_scan with who_is request to platform.bacnet_proxy
2020-11-04 21:16:25,505 (vcplatformagent-4.8 310) volttron.platform.agent.bacnet_proxy_reader INFO: Starting WHOIS
2020-11-04 21:16:25,511 (bacnet_proxyagent-0.5 307) __main__ DEBUG: Sending WhoIs: low_id: None high: None address: 10.10.220.101
2020-11-04 21:16:25,604 (bacnet_proxyagent-0.5 307) __main__ DEBUG: Calling IAm callback.
2020-11-04 21:16:25,605 (bacnet_proxyagent-0.5 307) __main__ DEBUG: IAm received: Address: 10.10.220.101 Device ID: 10012 Max APDU: 1024 Segmentation: segmentedBoth Vendor: 5
2020-11-04 21:16:25,608 (vcplatformagent-4.8 310) volttron.platform.agent.bacnet_proxy_reader INFO: Received iam for device_id 10012
2020-11-04 21:16:25,608 (vcplatformagent-4.8 310) volttron.platform.agent.bacnet_proxy_reader DEBUG: iam message is: {'address': '10.10.220.101', 'device_id': 10012, 'max_apdu_length': 1024, 'segmentation_supported': 'segmentedBoth', 'vendor_id': 5}
2020-11-04 21:16:25,608 (vcplatformagent-4.8 310) __main__ DEBUG: IAM response message: {'address': '10.10.220.101', 'device_id': 10012, 'max_apdu_length': 1024, 'segmentation_supported': 'segmentedBoth', 'vendor_id': 5}
2020-11-04 21:16:25,608 (vcplatformagent-4.8 310) __main__ INFO: IAM response for (andress, devices_id) (10.10.220.101, 10012)

[... cut ...] x7f735ea11b70>, {'': defaultdict(<class 'set'>, {'devices': {<bound method VolttronCentralPlatform._on_device_publish of <__main__.VolttronCentralPlatform object at 0x7f735e9edbe0>>}, 'status': {<function VCConnection.subscribe_to_vcp.<locals>.subscription_wrapper at 0x7f735d9c6f28>}, 'iam/': {<function VCConnection.subscribe_to_vcp.<locals>.subscription_wrapper at 0x7f735d9ce0d0>}, 'configure/': {<function VCConnection.subscribe_to_vcp.<locals>.subscription_wrapper at 0x7f735d9ce1e0>}})})})
```

acedrew, 2020-11-05:
Are you able to complete the scan using the command line interface?

gnmerritt, 2020-11-09:
Yep, works fine from there (and I get the same response that I'm seeing in the edge node logs)

```
$ python3 scripts/bacnet/bacnet_scan.py --ini ~/BACpypes.nathan.ini --debug --color --csv-out scan.csv --timeout 60 --address 10.10.220.101
$ cat scan.csv
address,device_id,max_apdu_length,segmentation_supported,vendor_id
10.10.220.101,10012,1024,segmentedBoth,5
```

acedrew, 2020-11-09:
I meant using the proxy scan, which is the same mechanism the VC process
would use.

Andrew

gnmerritt, 2020-11-09:
Ahh, that makes sense. Thanks for clarifying. 

Looks like that script doesn't work with python 3 (appears to be the same problem I fixed for the other script here https://github.com/VOLTTRON/volttron/pull/2469)

```
$ python3 scripts/bacnet/proxy_bacnet_scan.py --address 10.10.220.101 --timeout 60 --csv-out proxy-scan.csv
2020-11-09 16:49:41,041 __main__ ERROR: an error has occurred: %s
Traceback (most recent call last):
  File "scripts/bacnet/proxy_bacnet_scan.py", line 160, in <module>
    main()
  File "scripts/bacnet/proxy_bacnet_scan.py", line 129, in main
    csv_writer.writeheader()
  File "/usr/lib/python3.7/csv.py", line 144, in writeheader
    self.writerow(header)
  File "/usr/lib/python3.7/csv.py", line 155, in writerow
    return self.writer.writerow(self._dict_to_list(rowdict))
TypeError: a bytes-like object is required, not 'str'
```

But once I patch it (will put a PR up in a sec) it seems to be get as far as I was seeing in the logs: 

```
$ python3 scripts/bacnet/proxy_bacnet_scan.py --address 10.10.220.101 --timeout 60 --csv-out proxy-scan.csv
2020-11-09 16:53:45,800 root DEBUG: Creating ZMQ Core None
2020-11-09 16:53:45,802 volttron.platform.vip.zmq_connection DEBUG: ZMQ connection 242b2500-7bc1-4294-a653-96169510e83c
2020-11-09 16:53:45,804 volttron.platform.vip.zmq_connection DEBUG: connecting to url ipc://@/home/volttron/.volttron/run/vip.socket?publickey=sxXGsLEcJuqq2NY242jzGWQ1tInXTtHF6ds5j-pdkx4&secretkey=<redacted>
2020-11-09 16:53:45,804 volttron.platform.vip.zmq_connection DEBUG: url type is <class 'str'>
{'address': '10.10.220.101',
 'device_id': 10012,
 'max_apdu_length': 1024,
 'segmentation_supported': 'segmentedBoth',
 'vendor_id': 5}

$ cat proxy-scan.csv
address,device_id,max_apdu_length,segmentation_supported,vendor_id
10.10.220.101,10012,1024,segmentedBoth,5
```

And I see the following in the volttron logs:

```
2020-11-09 16:53:45,819 (bacnet_proxyagent-0.5 308) __main__ DEBUG: Sending WhoIs: low_id: None high: None address: 10.10.220.101
2020-11-09 16:53:45,905 (bacnet_proxyagent-0.5 308) __main__ DEBUG: Calling IAm callback.
2020-11-09 16:53:45,905 (bacnet_proxyagent-0.5 308) __main__ DEBUG: IAm received: Address: 10.10.220.101 Device ID: 10012 Max APDU: 1024 Segmentation: segmentedBoth Vendor: 5
```

acedrew, 2020-11-09:
Try running from VC with a much shorter duration, like 10 seconds.

gnmerritt, 2020-11-09:
No change from my first attempt, the UI sits and spins forever same as with the longer timeout. 

[image]

It seems to me like there may be a communication problem between the edge and VC nodes, but like I said before I'm not sure about the best way to tackle debugging. I would guess that it's unlikely that the message bus is totally broken given that I can both initiate the scan from VC and see which agents are running on the edge node.

acedrew, 2020-11-09:
Yes, the debugging step above is based on experience I've had where gevent timeouts aren't always aware of things like remote RPC timeouts and you can simply drop communications before a task finishes. Sounds like the issue may be a bit deeper. I don't use VC at all, so I'm a little blind on its inner workings, at least since a few versions ago.

bonicim, 2022-04-29:
Fixed in #2527

closed 2022-04-29 by bonicim