VOLTTRON/volttron #2566: Error

GeorgeIoak, 2021-01-18:
### Description of Issue
Trying to run through the initial install steps

### Affected Version
7.0
### Screenshots

#### Expected
Start the VOLTTRON instance
#### Actual
pika.adapters.base_connection ERROR: Read empty data, calling disconnect
### Steps to Reproduce
Step 4 at Test
```
(volttron) george@iotllc:~/volttron$ volttron -vv -l volttron.log &
[1] 2825
(volttron) george@iotllc:~/volttron$ Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 854, in gevent._gevent_cgreenlet.Greenlet.run
  File "/home/george/volttron/volttron/platform/vip/proxy_zmq_router.py", line 163, in on_stop
    self.zmq_router.stop()
  File "/home/george/volttron/volttron/platform/vip/router.py", line 131, in stop
    self.socket.close(linger)
AttributeError: 'NoneType' object has no attribute 'close'
2021-01-17T23:40:28Z <Greenlet at 0x7f9c6d7d0748: <bound method ZMQProxyRouter.on_stop of <volttron.platform.vip.proxy_zmq_router.ZMQProxyRouter object at 0x7f9c6d7e0e10>>(<volttron.platform.vip.agent.core.RMQCore object a)> failed with AttributeError
```

### Additional Details

[volttron.log](https://github.com/VOLTTRON/volttron/files/5827041/volttron.log)

- Log file is attached.
- I'm running Ubuntu 18.04 on a Digital Ocean Server
- my /etc/hosts file contains `127.0.0.1       localhost iotllc`
- I've tried both the main and the develop branch
- Python version is 3.6.9
- I get the same log if I use `./start-volttron`

craig8, 2021-01-25:
Hi @GeorgeIoak I have asked one of the team to take a look at this.  I see that you are using a provider that we haven't used.  Have you attempted this locally within a virtual environment or linux system?

bonicim, 2021-01-25:
Hi @GeorgeIoak , I've tried to reproduce the error on my end but to no success. Note that I do not have a Digital Ocean Server so I had to reproduce this on my VM (i.e. Virtual Box). To setup volttron, I did a fresh install of volttron (`python bootstrap.py --force`), ran the RMQ dependencies script (`./scripts/rabbit_dependencies.sh debian`), activated the virtual environment (`source env/bin/activate`), configured the platform to use RMQ (`python bootstrap.py --rabbitmq single`), and then started volttron (`volttron -vv`).  I did read the attached logs and saw that there were some RMQ connection errors. One follow up questions for you: Did you use `python bootstrap.py --rabbitmq single` to install RabbitMQ before starting volttron?

GeorgeIoak, 2021-01-26:
Hi @bonicim , thank for looking into this. Virtual Box was my first choice to try this out but unfortunately I run Windows 10 on my main desktop and my VMs are currently not working.

I had an old unused instance at Digital Ocean so I thought I'd try it but it did need some severe upgrading so something there could have gone wrong. Your steps look familiar except for adding `single` for RMQ.

I did notice the RMQ errors and wondered if they had anything to do with using localhost instead of a FQDN?

I'll go back and wipe out that install of volttron and try it again and report back if I get the same error or not.