VOLTTRON/volttron #2596: 8.0 Release won't start without RabbitMQ dependencies, even using ZMQ

acedrew, 2021-02-04:
### Description of Issue
A fresh install of 8.0 will not start without installing the RabbitMQ Server and all of it's dependencies
### Affected Version
8.0
### Screenshots

#### Expected

#### Actual
Fails with assertion error that the rabbitmq server directory does not exist

### Steps to Reproduce

### Additional Details

shwethanidd, 2021-02-04:
@acedrew: working on it now.

jhaack, 2021-02-04:
Stacktrack:
2021-02-04 09:12:19,543 () volttron.platform.main ERROR: Missing rabbitmq server directory/home/jer/rabbitmq_server/rabbitmq_server-3.7.7/
Traceback (most recent call last):
  File "/home/jer/git/vdev/volttron/platform/main.py", line 1094, in start_volttron_process
    health_service = HealthService(address=address,
  File "/home/jer/git/vdev/volttron/platform/vip/healthservice.py", line 63, in __init__
    self._rmq_config = RMQConfig()
  File "/home/jer/git/vdev/volttron/utils/rmq_config_params.py", line 94, in __init__
    assert os.path.isdir(self.rabbitmq_server), "Missing rabbitmq server directory{}".format(self.rabbitmq_server)
AssertionError: Missing rabbitmq server directory/home/jer/rabbitmq_server/rabbitmq_server-3.7.7/
2021-02-04 09:12:19,545 () volttron.platform.main ERROR: None

jhaack, 2021-02-04:
Creating the following directory fixes the problem:
/home/jer/rabbitmq_server/rabbitmq_server-3.7.7/

The check for this directory needs to be removed for ZMQ installs. Tests need to be adjusted to catch this error.

jhaack, 2021-02-07:
Thank you, @acedrew. The release has been pulled back and will be republished after additional tests are in place.

shwethanidd, 2021-03-12:
@acedrew Please test releases/8.x branch and close the issue

PR #2597 (2021-02-05): Fixes Issue#2596. RMQ config object was created without checking for …
# Description
RMQ config object was created without checking for message bus flag

Fixes #2596
closed 2021-03-19 by shwethanidd