VOLTTRON/volttron #2607: ForwardHistorian in bad state after losing connectivity to target platform

acedrew, 2021-02-12:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue

### Affected Version
Volttron 6.x Forwarder 4.0

### Screenshots

"Feb  9 13:56:06 vtron-node 2021-02-09 13:56:06,059 (forwarderagent-4.0 22260) forwarder.agent ERROR: Traceback (most recent call last):#012  File "/home/volttron/.volttron/agents/35317bb8-fd76-47fa-b767-6713f4d908c7/forwarderagent-4.0/forwarder/agent.py", line 361, in publish_to_historian#012    message=payload['message']).get()#012  File "/var/lib/volttron/env/local/lib/python2.7/site-packages/gevent/event.py", line 373, in get#012    return self.get(block=False)#012  File "/var/lib/volttron/env/local/lib/python2.7/site-packages/gevent/event.py", line 363, in get#012    return self._raise_exception()#012  File "/var/lib/volttron/env/local/lib/python2.7/site-packages/gevent/event.py", line 343, in _raise_exception#012    reraise(*self.exc_info)#012  File "/var/lib/volttron/env/local/lib/python2.7/site-packages/gevent/_util_py2.py", line 8, in reraise#012    raise type, value, tb#012Again: VIP error (11): Resource temporarily unavailable"

#### Expected
ForwardHistorian caches data until connectivity is restored, then delivers data to the target platform.

#### Actual
Delivery is intermittent and agent reports bad state while continuing to cache data.

### Steps to Reproduce

### Additional Details

acedrew, 2021-02-14:
@mdermond

acedrew, 2021-03-26:
I have this happening in 7.x under python 3 now as well. I've dug in a bit, it looks like there is maybe some bad state left around in the vip connection? The error ends up coming from volttron.platform.vip.agent.core, there's no exception as above in the 7.x code, but the errors continue until the FH is restarted.

jhaack, 2021-03-26:
@craig8  If this can't be resolved another way, could this be a case where the agent restarter might be worth investigating as a fallback? It would at least automatically handle this case.

acedrew, 2021-03-26:
Can't post logs here, but happy to review with @craig8 or anyone else synchronously

craig8, 2021-03-26:
@jhaack this is exactly that case...a small amount of difference than what Jamie's does but I can certainly see if I can get this out for the public....@mdermond