VOLTTRON/volttron #2623: VCTL status does not display UUID column header

jklarson, 2021-02-24:
### Description of Issue

The VCTL status command does not display 'UUID' column header.  This is confusing for users.

#### Expected

UUID  AGENT             IDENTITY            TAG       STATUS          HEALTH
a listeneragent-3.3 listeneragent-3.3_2 listener2 running [12872] GOOD
6 listeneragent-3.3 listeneragent-3.3_1 listener1 running [12873] GOOD

#### Actual

  AGENT             IDENTITY            TAG       STATUS          HEALTH
a listeneragent-3.3 listeneragent-3.3_2 listener2 running [12872] GOOD
6 listeneragent-3.3 listeneragent-3.3_1 listener1 running [12873] GOOD

craig8, 2021-06-05:
@bonicim I am going to work on this as I have a bunch of changes to vctl that this would go along with.  #2579

craig8, 2021-06-05:
#2688

bonicim, 2021-06-07:
> @bonicim I am going to work on this as I have a bunch of changes to vctl that this would go along with. #2579

Ok.