VOLTTRON/volttron #2654: BACnet script grab_multiple_configs.py will hang when device communication is not established.

rlutes, 2021-04-01:
NOTE issues are for bugs/feature requests not support.  Please ask at https://stackoverflow.com/questions/tagged/volttron for help

### Description of Issue
When running this script on a large number of devices the script will hang if connection to one of the devices fails.

### Affected Version
releases/8.x
### Screenshots

#### Expected
Script should move to next device.  Ideally a list of failed devices might be generated.
#### Actual

### Steps to Reproduce

### Additional Details

craig8, 2021-04-01:
@rlutes  can you remove the boiler plate from your issue.  Provide an example as well.

craig8, 2021-04-01:
@jklarson is this similar or not to https://stackoverflow.com/questions/66889829/volttron-nosegmentationsupported-for-bacnet-devices