VOLTTRON/volttron #2660: Config store commands hang occasionally with PlatformDriverAgent

jklarson, 2021-04-15:
### Description of Issue

On occasion, config store commands will hang after having completed their intended work (i.e. if running vctl config store, the config will be stored in the right place, however the command will not exit until timing out).  I have yet to determine an underlying thread, but the behavior does appear to in some way be sporadic.

### Affected Version

8.x

#### Expected

Config store commands should exit gracefully after completing their respective tasks.

#### Actual

Config store commands hang after doing their work and finally close after the timeout period (~30 seconds)

### Steps to Reproduce

Store a configuration file for the platform driver - I have seen this behavior multiple times, but it does not appear to be something easy to reproduce from a clean install.

rlutes, 2021-04-15:
There can also be some odd behavior when using the vctl config edit.  The shell session will freeze up and hang and one will have to close and reopen another session.

craig8, 2021-04-16:
So this is due to a gevent timeout.  You only have a specific time period to edit the file.  We could increase it to whatever, but you would still get the same behavior if you didn't finish with-in that time....