VOLTTRON/volttron #2674: rotatinglog.py

bbartling, 2021-05-03:
### Description of Issue
Hello I posted a question on Stackoverflow for a [volttron log file management best practices](https://stackoverflow.com/questions/67338712/volttron-log-file-management-best-practices) and perhaps I didnt understand the answer correctly.

While VOLTTRON was running I entered the answer into the terminal that was posted on SO. I think @craig8 was trying to point to me on SO this is written into the bash script to `./Start-Voltton` but don't actually run it in terminal.

Not really knowing I ran this in terminal during a running session and it seemed to kill or disconnect the VOLTTRON pid:
`volttron -L examples/rotatinglog.py > volttron.log 2>&1 &`

At least once that is ran on a running session, `vctl status` is gone, its like VOLTTRON is no longer running but I could tell data was still coming through to our Grafana interface. 

Basically to get VOLTTRON back I had to do a `killall volttron` and then restart everything in the root user. Hopefully that makes sense. Thanks  @acedrew  for tips as well.

### Affected Version
VOLTTRON 8.0

bbartling, 2021-05-03:
`./start-volttron --rotating` is the proper methods to use rotating log files?

acedrew, 2021-05-03:
Just to clarify, it really has nothing to do with rotating log, it's attempting to run /env/bin/volttron while there is already a background process fails and cleans up the PID or socket?, preventing either stop-volttron.sh or /env/bin/vctl from working

craig8, 2021-05-03:
yeah the ./start-volttron does not check for existing running volttrons.  Andrew's comment is 100% correct.  The logging has nothing to do with this issue.

acedrew, 2021-05-03:
@craig8 It's not really ./start-volttron's fault either, as he didn't run start-volttron, he ran the bare executable /env/bin/volttron
The auto cleanup must be built into the core platform, which it should do on error, but only if it wasn't already present when it attempted to start.