VOLTTRON/volttron #2689: Volttron Not Starting

Shalomated, 2021-06-01:
Hello, please I am new to volttron. After installing and activating using "source env/bin/activate", I tried starting volttron using ./start-volttron, but get the feed back below

[image]

On checking the volttron.log file, this is the output

[image]

Please help

craig8, 2021-06-01:
> Check twice before you click! This email originated from outside PNNL.

Based upon what I see it means there are more than one volttrons running.

Please execute:

ps aux |grep volttron

If there are volttrons there then

killall -9 volttron

That should fix the issue....

[image]<https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F85184890%2F120357744-62ff9a00-c2cb-11eb-864d-a1431bb98964.png&data=04%7C01%7CCraig.Allwardt%40pnnl.gov%7Cfad9085b8fc64e8bffa708d92519ac54%7Cd6faa5f90ae240338c0130048a38deeb%7C0%7C0%7C637581614537231627%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=UQtrE3m%2BTIiJBFpyhSGOrUBNkcPb%2F8GfeDspyAIfJYE%3D&reserved=0>

[image]<https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F85184890%2F120358066-bb369c00-c2cb-11eb-935d-509b697e37a7.png&data=04%7C01%7CCraig.Allwardt%40pnnl.gov%7Cfad9085b8fc64e8bffa708d92519ac54%7Cd6faa5f90ae240338c0130048a38deeb%7C0%7C0%7C637581614538057993%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=NLL3aU8i7VPrEF4LCj%2FznP%2FyrWMFtPBLE8a0YdepqhY%3D&reserved=0>

craig8, 2021-06-01:
@VOLTTRON/main-developers  we should see about making this a bit more robust when there is a bind error.  The only issue I am not sure about is if the arguments to start the agent aren't the same then what do we do?

Shalomated, 2021-06-01:
Dear Craig, thanks for the response. After trying out the command you suggested, this is the feedback I got

[image]

I am not sure, but from what I suspect, there are no volttrons as I newly installed volttron on this device

craig8, 2021-06-01:
killall -9 volttron

killall is one word.

Shalomated, 2021-06-02:
It worked!

Thanks a lot Craig

closed 2021-06-02 by craig8